pybricks/support #512: Can't update fw on CONTROL+ hub

DanieleBenedettelli, 2021-10-20:
**Describe the bug**
I follow these steps https://pybricks.com/install/technic-boost-city/
but this error appears ```
GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
```

**To reproduce**
follow the steps in the guide linked above 

4. See error

**Expected behavior**
fw download should begin

**Screenshots**
![error](https://user-images.githubusercontent.com/47357389/138075142-c6006881-92a3-4a7e-95e3-b9454856cae4.png)

dlech, 2021-10-20:
What operating system do you have?

Also see https://github.com/pybricks/support/discussions/270 for troubleshooting tips.

DanieleBenedettelli, 2021-10-20:
On Windows 10, no progress is shown. On two Android devices, it flashes
until 100%, but hub remains in bootloader mode