pybricks/support #513: Windows 10 can't connect to CONTROL+ hub

DanieleBenedettelli, 2021-10-20:
**Describe the bug**
I tried connecting to a CONTROL+ hub with 
code.pybricks.com and beta.pybricks.com 
from Chrome Versione 94.0.4606.81
on Windows 10 20H2 19042.1288
The hub had the fw installed using Android, because on this machine the fw upload failed.

**To reproduce**
Click on connect, select hub, wait

**Expected behavior**
The computer won't connect to the hub. On the Console, this error appears
DOMException: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`

**Screenshots**
Here are the logs from **code** and **beta**
[beta.pybricks.com-1634757617600.log](https://github.com/pybricks/support/files/7383781/beta.pybricks.com-1634757617600.log)
[code.pybricks.com-1634757458917.log](https://github.com/pybricks/support/files/7383794/code.pybricks.com-1634757458917.log)

dlech, 2021-10-20:
This sounds like #312. Can you try capturing Bluetooth packets to see if we see the same thing? 

Instructions: https://bleak.readthedocs.io/en/latest/troubleshooting.html#windows-10

DanieleBenedettelli, 2021-11-15:
@dlech I am trying flashing the firmware from an Android smartphone, it's the 3rd attempt, and when it arrives at 100%, it gives the error "The hub took too long to respond. Restart the hub and try again." 

The hub remains in bootloader mode when I power cycle it, and I have to use the LEGO Powered Up app to flash the original fw back. I really need the Technic hub for a commission. 
Two out of two devices won't flash it. Can it depend from the hub itself?

DanieleBenedettelli, 2021-11-15:
Can I use the BlueGiga BLED112 as dongle alternative to my PC embedded antenna?

Thanks,
D

dlech, 2021-11-15:
>  I am trying flashing the firmware from an Android smartphone,

Are you using https://beta.pybricks.com? There is a know issue with android on https://code.pybricks.com.

dlech, 2021-11-15:
> Can I use the BlueGiga BLED112 as dongle alternative to my PC embedded antenna?

If it has a Windows driver, it could work (you will need to disable the built-in Bluetooth device in *Device Manager*). I don't know if the problem you are having is with the computer or with the Bluetooth chip.

DanieleBenedettelli, 2021-11-15:
Shoot, I forget using beta! Let me try again from my 3rd device, and Android tablet.

DanieleBenedettelli, 2021-11-15:
Fw update succeeded from LENOVO tablet with beta. My PC cannot even connect when the fw is PyBricks to develop code.
I will try using Android

dlech, 2023-03-23:
Did you ever find a compatible adapter?

DanieleBenedettelli, 2023-03-23:
Yes, I bought the TP-LINK Bluetooth Adapter 5.0 UB 500 
https://amzn.to/3TFxC6A which works wonderfully!

dlech, 2023-03-23:
Awesome! Out of curiosity, what kind is the internal radio that doesn't work?

closed 2023-03-23 by dlech

DanieleBenedettelli, 2023-03-24:
My DELL laptop has a Dell Wireless 1820A Bluetooth 4.1 LE adapter (Broadcom Corporation)