pybricks/support #521: How do you create and use a sensor device for MicroInfinity CruizCore XG1300L in micropython?

LeeJenkins, 2021-10-30:
**Question**

How do you create and use a sensor device for MicroInfinity CruizCore XG1300L in micropython?

**Background Information**
I have a XG1300L plugged into port 3. I have a simple python3 program that successfully uses the sensor and prints angle values.

**Now I want to use the sensor in a micropython program, but I cannot get it working.**

**Screenshots**

The python3 program (working):
```python
#!/usr/bin/python3

from ev3dev2.sensor import *
import time

# in the ev3dev brickman interface, navigate to
#     Device Browser > Sensors
# there you will find an entry for the mi-xg1300l
# create the Sensor with the string reported there.
uis = Sensor("i2c-legoev35:i2c1")

print("uis.address",uis.address)
print("uis.bin_data_format",uis.bin_data_format)
print("uis.decimals",uis.decimals)
print("uis.driver_name",uis.driver_name)
print("uis.modes",uis.modes)
print("uis.mode",uis.mode)
print("uis.units",uis.units)

for i in range(33):
    print( uis.value(0) / 100 )
    time.sleep(1)
```

Micropython program using I2CDevice. https://docs.pybricks.com/en/stable/iodevices/i2cdevice.html
The program generates "OSError: [Errno 5] EIO" on the call to device.read():
```python
#!/usr/bin/env pybricks-micropython
from pybricks.hubs import EV3Brick
from pybricks.iodevices import I2CDevice
from pybricks.parameters import Port

# Initialize the EV3
ev3 = EV3Brick()

# Initialize I2C Sensor
CruizCore_XG1300L_I2C_address    = 0x02
CruizCore_XG1300L_read_angle_register = 0x42
GyroPort = Port.S3
print( "create an I2C device on port",GyroPort)
print( "    at I2C address %x"%CruizCore_XG1300L_I2C_address )
device = I2CDevice( GyroPort, CruizCore_XG1300L_I2C_address >> 1 )

print( "read one byte from register %x"%CruizCore_XG1300L_read_angle_register )
print( device.read( CruizCore_XG1300L_read_angle_register, 1 ) )
```

Micropython program using Ev3devSensor. https://docs.pybricks.com/en/stable/iodevices/ev3devsensor.html
The program generates "OSError: [Errno 5] EIO" on the call to Ev3devSensor():
```python
#!/usr/bin/env pybricks-micropython
from pybricks.hubs import EV3Brick
from pybricks.parameters import Port
from pybricks.iodevices import Ev3devSensor

# Initialize the EV3
ev3 = EV3Brick()

# Initialize Sensor
port = Port.S3
print( "create Ev3devSensor on",port )
sensor = Ev3devSensor( port )

print( "read sensor angle" )
angle = sensor.read('ANGLE')
print(angle)
```

**Desktop (please complete the following information):**
 - OS and version: Window 10
 - Connection Type: WiFi connection to EV3
 - Software: edit/run files over ssh connection

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: 
```
$ uname -a
Linux tiger 4.14.117-ev3dev-2.3.5-ev3 #1 PREEMPT Sat Mar 7 12:54:39 CST 2020 armv5tejl GNU/Linux
```

Please paste the system info from the ev3dev VS Code extension or the output of `ev3dev-sysinfo -m` here!

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3-micropython-v2.0.0-sd-card-image
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:          4.0
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016535492B5
BOARD_INFO_TYPE=main
```

dlech, 2021-10-30:
The `Ev3devDevice` case looks like a bug in Pybricks. 3rd party I2C sensors are excluded from `Ev3devDevice` ([source](https://github.com/pybricks/pybricks-micropython/blob/1473c4ba649fd38d07731c6104f66bf43b083d38/lib/ev3dev/src/ev3dev_stretch/lego_sensor.c#L84-L129)). Otherwise that would be the ideal way to use it.

For the `I2CDevice` case, you probably need to go through the Brickman menu and manually change input port 3 to `other-i2c` before running the program.

LeeJenkins, 2021-10-30:
I changed the input type to `other-i2c` and re-ran the I2CDevice program, but the result was the same.

So are 3rd party sensors (and this one specifically) excluded as a matter of policy, or just because no one has the sensor for testing? If it's the former, then the fix requires adding another else-if with the xg1300l driver and rebuilding, right? If it's the latter, then I would volunteer to test the code change. Then others could benefit from the change as well. :-)

laurensvalk, 2021-10-30:
> So are 3rd party sensors (and this one specifically) excluded as a matter of policy?

No, that’s exactly what the Ev3devSensor class is intended for. It has been tested with several third party I2C sensors at the time. This class specifically does not check device ID and is supposed to support anything that ev3dev supports, but apparently something is wrong here.

It’s been a while since I tried running anything on the EV3. Maybe I can try to have a look next week :)

dlech, 2021-10-30:
> I changed the input type to `other-i2c` and re-ran the I2CDevice program, but the result was the same.

Unfortunately, all errors are currently mapped to EIO, so it is very difficult to try to guess what the problem may be.

> So are 3rd party sensors (and this one specifically) excluded as a matter of policy,

I think it was just overlooked or not prioritized. 

>  If it's the former, then the fix requires adding another else-if with the xg1300l driver and rebuilding, right?

There was supposed to be a `/sys/bus/i2c/drivers/nxt-i2c-sensor/driver_names` attribute like there is for other types of sensors (analog, etc.) that would have provided all possible I2C driver names at runtime, but it looks like this wasn't implemented in ev3dev. So instead, we will need to add all of [these](https://github.com/ev3dev/lego-linux-drivers/blob/2c05d904e82ee62a5425ba244256a9cf7928f5a2/sensors/nxt_i2c_sensor.h#L167-L198) minus the official LEGO devices that are already detected. Instead of adding an `else if` for each one, we can just make a list iterate it. If anything matches, we can set the ID to `PBIO_IODEV_TYPE_ID_NXT_I2C` (instead of making a new ID for each sensor). And it might not hurt to add

        (id == PBIO_IODEV_TYPE_ID_NXT_I2C && status_now == NXT_I2C) ||

to [here](https://github.com/pybricks/pybricks-micropython/blob/1473c4ba649fd38d07731c6104f66bf43b083d38/lib/ev3dev/src/ev3dev_stretch/lego_port.c#L139-L147) to prevent the port mode from being switched back to `auto` (I think there is one I2C sensor that doesn't work with `auto` - don't remember which one).

dlech, 2021-10-30:
Laurens wrote the code originally, so he should know best. Maybe what I suggested is not necessary.

LeeJenkins, 2021-11-01:
> It’s been a while since I tried running anything on the EV3. Maybe I can try to have a look next week :)

Thanks!

laurensvalk, 2021-11-02:
Based on your first post, I'm going to guess that the `address` is something like:

``i2c-legoev35:i2c1``

There does not seem to be any port information there, so Pybricks has no way of knowing this is `Port.S3`. I'm not sure exactly what this address string means in ev3dev.

LeeJenkins, 2021-11-06:
By the way, this sensor requires an initialization step once per boot. The ev3dev programs cannot utilize the sensor (there is no directory /sys/class/lego-sensor/sensorX) unless it has been initialized. See see https://github.com/ev3dev/ev3dev/issues/1011. I've put the initialization into a script:

```
#!/bin/sh
# the i2c string below uses the port number + 2
# e.g. if the sensor is plugged into port 4, then
# the string should be i2c-6
sudo sh -c 'echo "mi-xg1300l 0x01" > /sys/bus/i2c/devices/i2c-6/new_device'
```

Here's the address and driver name information:

```
robot@tiger:/sys/class/lego-sensor/sensor1$ cat address
i2c-legoev36:i2c1
robot@tiger:/sys/class/lego-sensor/sensor1$ cat driver_name
mi-xg1300l
```

I was curious that the address doesn't match the string I used in my test script (it says 36 now instead of 35). But I changed the port from S3 to S4, so that part of the address changed to 36 instead of 35. So that digit matches the port-number+2 digit in the init script.

Here's the python3 program output:

```
robot@tiger:~/lab-test$ ./microinfinity-py3-working.py
uis.address i2c-legoev36:i2c1
uis.bin_data_format s16
uis.decimals 2
uis.driver_name mi-xg1300l
uis.modes ['ANGLE', 'SPEED', 'ACCEL', 'ALL']
uis.mode ANGLE
uis.units deg
0.0
7.11
18.75
29.87
51.07
74.59
23.88
-21.99
... etc.
```

Does that help?

laurensvalk, 2021-11-06:
Thanks for the info. So if the address can consistently be mapped to a port like you described, the fix could be to generalize this line:

https://github.com/pybricks/pybricks-micropython/blob/c556d1df1e41652f61c2261fb34984a1f9e28bae/lib/ev3dev/src/ev3dev_stretch/sysfs.c#L44

to get the correct port from that form of address string as well.

We’re still trying to figure out if we should continue supporting EV3 with new updates. What do you think? 😄

LeeJenkins, 2021-11-06:
Ha. :-) Of course I'm biased because my team is sitting on a pile of EV3s. 

Regarding the line of code, it looks like it is reading the sysfs file **/sys/class/lego-sensor/sensorN/address**, right? So are you saying that the general solution is to change the format of the address reported through sysfs? Would that require a change in the driver code? Or am I over-thinking this?

It might be possible to localize the change to the pybricks code with a second fscanf that only happens if the first one fails.

```
            char port_char;
            if (fscanf(f_address, "ev3-ports:%*[a-z]%c", &port_char) < 1) {
                rewind(f_address);
                if (fscanf(f_address, "i2c-legoev3%[3-6]:", &port_char) == 1) {
                    port_char -= 2;
                }
                else {
                    return PBIO_ERROR_IO;
                }
            }
```

Which approach sounds better to you from a support standpoint?

LeeJenkins, 2021-11-07:
@laurensvalk  While looking at this again this morning, I just realized that the pybricks code has memory leaks with the return statements that occur while the directory and the file are open for reading. Just an observation.

laurensvalk, 2021-11-07:
> So are you saying that the general solution is to change the format of the address reported through sysfs? 

No, just generalize how it's read in Pybricks so we cover all cases. It looks like you've already figured it out :+1:

> I just realized that the pybricks code has memory leaks with the return statements that occur while the directory and the file are open for reading. Just an observation.

Thanks. If we're going to continue EV3 support as-is, much of this pybricks-ev3dev glue code could certainly use some work. There's been some ideas to proceed instead with a "light" version of ev3dev, but there aren't any specific plans as we've been working on the newer hubs.

There's even a proof of concept of running Pybricks on the good-old NXT (it works, but most drivers are still missing) if you're looking for another challenge :wink: .

laurensvalk, 2025-07-13:
Although the original question remains valid and probably will be supported too, via the `iodevices.I2CDevice` class.

laurensvalk, 2026-10-08:
5 years since OP @LeeJenkins opened this, it's worth adding that this is now easy to do on the brand new [Pybricks for EV3](https://beta.pybricks.com/)!

Here's an example of making your own class to wrap reading from certain registers and returning a useful numeric format. When done like this, the whole class is also async-compatible.

```python
class MyUltrasonic:

    def __init__(self):
        self.i2c = I2CDevice(Port.S4, address=0x01)

    def distance(self):
        return self.i2c.read(0x42, 1, lambda data: data[0] * 10)

sensor = MyUltrasonic()
print(sensor.distance())
```

We'd love to see this in action on some real devices like yours.