pybricks/support #523: Freeing up more space on Move hub

dlech, 2021-11-08:
Move hub is once again dangerously close (less than 500 bytes) to running out of flash memory for the firmware. So, we need to come up with more ideas to free up space.

dlech, 2021-11-08:
Idea: disable WeDo 2.0 and/or SPIKE I/O devices.

The recent `ColorLightMatrix` class added 500 bytes. So if each of these device is in that ballpark, this could potentially save 2-3k.

laurensvalk, 2021-11-08:
https://github.com/pybricks/pybricks-micropython/commit/7dd84a3 seems much larger than I expected, so I'll need to have a look at that one.

laurensvalk, 2021-11-09:
> pybricks/pybricks-micropython@7dd84a3 seems much larger (+1K) than I expected

Managed to cut it by 508 bytes. Combined with your 320 byte savings for the firmware info, we might have a little longer to go.

laurensvalk, 2021-11-18:
Idea: disable the `Control()` class on motors and drive bases.

Really, the defaults will do fine for most cases.

One downside is that `control.stalled` and `control.done` would no longer be available so multi-motor control would be a bit harder. These are already being flattened in the `DriveBase` case, so maybe it makes sense to do that for `Motor` as well.

laurensvalk, 2021-12-21:
Disabling `Motor.control`, `Motor.log`, and the corresponding control and log objects in `DriveBase` saves about 2800 bytes, which should keep us going for a little while longer.

dlech, 2021-12-22:
FYI, https://github.com/pybricks/pybricks-micropython/commit/44ee268a37aa8e87e13ef17c9eddd1676bfbe8c6 breaks our [R2-D2 example program](https://pybricks.com/projects/sets/boost/droid-commander/r2-d2/) since is uses `drive.distance_control.limits()`.

laurensvalk, 2021-12-22:
I’d need to have a closer look tomorrow, but I think the limits() call here has become redundant anyway. Just the settings() call should suffice, and we’re keeping that one.

c59-zurich, 2021-12-29:
Hello, thank you for developing this very useful tool!
I'm new to the community and not a coding expert. What if, like for the option to download the program together with the firmware, there is an option on which firmware blocks to include in the firmware to download? This way you could save memory when downloading not needed firmware blocks knowing of course that the developing platform is not complete anymore for new developments....
Is there also a way to make a program autostart just to start the downloaded program right when powering up?
This way you could make "permanent final programs" which uses less memory as they are not intended for development anymore.
If you need do develop a new program then you reinstall the full firmware and start the program with the play icon.
Even nicer would it be if you could determine which firmware blocks to download automatically by downloading of the firmware+program operation.
Kind regards

dlech, 2021-12-29:
> What if, like for the option to download the program together with the firmware, there is an option on which firmware blocks to include in the firmware to download?

This is probably too complicated. We would have to build a separate firmware for each combination of features. Pro users can download the source code and do this themselves though.

> Is there also a way to make a program autostart just to start the downloaded program right when powering up

No, this isn't possible. However, it only takes one extra press of the button to start the program.

c59-zurich, 2022-01-12:
The problems with the poor lego BOOST move hub memory are really limiting the possibility of implementing meaningful programs with this device, especially when considering a pybricks firmware that supports all possible devices. If considering reducing the amount of supported peripheric devices there seem not to be a Tecnical objective choice on which peripheric device to support and which not. As the lego BOOST move hub memory is really limited my suggestion is to limit the supported peripheric devices to the standard lego BOOST set at least to give a consistency on the purchased package which would probably fits to the majority of those who purchased this lego set and giving them the possibility to make meaningful programs.
Kind regards

dlech, 2022-10-10:
We have made good progress on freeing up space. There is a more general discussion about the future of move hub at #218.

closed 2022-10-10 by dlech