pybricks/support #537: Lego move hub   -   insufficient flash memory

MaxBrille, 2021-11-21:
Trying to load my Python code down to the move hub I get the message: 
'The combined firmware and main.py are too big to fit in the flash memory.' 

I saved my Python code via Thonny. The size of the file is less than 3 kB! 
So what is the real problem? Is there any workaround to get my code dowloaded? 

Anyway you are doing a great job! Thanks very much.

dlech, 2021-11-21:
The Move hub currently (v3.1.0rc1) has less than 1k free for the user program. See #523 and #218. The problem is that it only has 128kB total flash compared to 256kB (or more) on other hubs.

There is an unused 2k section of the flash memory beyond the space allocated for firmware that we might be able to take advantage of to allow slightly larger user programs.

laurensvalk, 2021-11-22:
In fact, it has so little space, that we've been wondering whether we should keep supporting it. But as long as Move Hub users like yourself keep telling is that they're using it, we probably will for a while longer :)

Do you want to share your code here? We might be able to give you some tips to optimize.

MaxBrille, 2021-11-22:
Thank you for the responses. 
It's really a pity. But of course you are truly nor responsible for the poor flash memory size in the move hub. 
As requested I'm attaching the code file to this email. 
You will realize that the actual status is just the beginning of my project 
(please refer to the second program loop, which represents the autoMode). 
Regarding the problems I think it is sadly senseless to keep on.
[2021-11-21 MoveHub.zip](https://github.com/pybricks/support/files/7579661/2021-11-21.MoveHub.zip)

MaxBrille, 2021-11-22:
Yes, you are right. 
The problem occurs only if I'm trying to save the program on the move hub. 
Starting the program on my PC does work.

laurensvalk, 2021-11-22:
This isn't going to save the day, though while we're here:

```python
        if speedA > 100:
            speedA = 100
        if speedA < -100:
            speedA = -100
        if speedB > 100:
            speedB = 100
        if speedB < -100:
            speedB = -100

        motorA.dc(-speedA)
        motorB.dc(speedB)
```

The `dc` method caps it to 100 internally anyway, so you can save some code size by dropping this check.

You may also be able to put your button/actions together in a single dictionary, but I'm not sure how big the savings will be, if any.

laurensvalk, 2021-11-22:
> There is an unused 2k section of the flash memory beyond the space allocated for firmware that we might be able to take advantage of to allow slightly larger user programs.

Since V3.1 might be the final release for Move Hub, this idea has my vote if there are no technical objections :+1:

MaxBrille, 2021-11-22:
I reduded the code as suggested but, I'm afraid, without success.

MaxBrille, 2021-11-22:
After reducing the program to the first loop (manual mode), download is possible and the move hub is working. 
The reduced program code consists of 78 lines instead of 114 lines (including blank lines). 
The restriction related to the flash memory size in the move hub is a serious matter. 
[2021-11-22 MoveHub.zip](https://github.com/pybricks/support/files/7580801/2021-11-22.MoveHub.zip)

laurensvalk, 2021-11-22:
Yes, hence the suggestion to either add those 2K of reserve space. Right now, there is only about 1K. So, that would be 3 times as much as now.

That is still limited, but much less so. And it couldn't be a whole lot more anyway, as the scripts are loaded into RAM and that's severely limited on Move Hub as well.

MaxBrille, 2021-11-23:
Regarding the memory problem of the MoveHub I'm thinking about switching to another LEGO hub. 
May I kindly ask if there is an overview sheet available concerning flash memory and RAM in the various hubs and the amount of free space for programs? Thank you in advance.

laurensvalk, 2021-11-23:
Good point. I think we could add that to [this table](https://github.com/pybricks/technical-info/blob/master/hardware/programmable-bricks.md), which already compares most other features.

In order of performance, you have:

- Move Hub (106K firmware flash, 16K RAM)
- City Hub (232K: firmware flash, 32K RAM)
- Technic Hub (232K: firmware flash, 64K RAM)
- SPIKE Essential Hub (1M: firmware flash, ?? M user flash, 256K RAM)
- SPIKE Prime Hub / MINDSTORMS Robot Inventor Hub (1M: firmware flash, 32 M user flash, 256K RAM)

BertLindeman, 2021-11-23:
MaxBrille: Do not forget to take the number of Powered Up ports into account.
In the table Laurens mentionned above as `Powered Up I/O connector`

Laurens: Should we add "the MoveHub has two builtin motors" to that table?

MaxBrille, 2021-11-23:
In addition to these data I found on wikipedia 'Lego Mindstorms' (unfortunately on the German web page) some data concerning display and processor: https://de.wikipedia.org/wiki/Lego_Mindstorms
[image]

laurensvalk, 2021-11-23:
All of the required information can be dug out of our configuration and linker files. We'll take a pull request from anyone who wants to update the table in our technical-info repo. Otherwise, we'll do it ourselves eventually :smile:

laurensvalk, 2022-01-21:
@MaxBrille: We have recently freed up quite a bit of space on Move Hub.

If you want to try it before we make a new release, you can use [these instructions](https://pybricks.com/install/technic-boost-city/#installing-the-latest-build-advanced).

Since we have a dedicated issue to free up even more space in #523, we can close this one.

closed 2022-01-21 by laurensvalk