pybricks/support #547: Cannot end the while loop on MoveHub

MarshToyokan, 2021-11-25:
I am currently working on a line following and writing a code for turning left at a crossroad. I used its colour sensor (as it should be) to detect the white colour while moving at 90 degs until it finds the ground white. But it just keeps moving around and won't stop at all. I believe there's something wrong with my codes but I am not sure which doesn't work well because I am doing this in a class and I started python just a few days ago. Sorry for my bad English, I am reading an instruction that my teacher gave us but I just cannot understand some parts. Thank you.

```
from pybricks.hubs import MoveHub
from pybricks.pupdevices import Motor, ColorDistanceSensor
from pybricks.parameters import Port, Direction
from pybricks.tools import wait, StopWatch
from pybricks.robotics import DriveBase

class LineFollower(DriveBase):
    
    def __init__(self, *args, **kwargs):
        super().__init__(*args, **kwargs)
        self.sensor = ColorDistanceSensor(Port.D)
        self.hub = MoveHub()
        print('Voltage {} mV'.format(self.hub.battery.voltage()))
        
    def run(self, min_runtime=1000, max_runtime=30000, max_duration=100):
        watch_runtime = StopWatch()
        watch_runtime.reset()
        watch_online = StopWatch()
        watch_online.reset()
        
        print('following line...')
        
        WHITE = 20
        LIGHTGREY = 14
        DARKGREY = 10
        BLACK = 7
        
        while watch_runtime.time() < max_runtime:
        
            intensity = self.sensor.reflection()
            
            
            if intensity > WHITE:
                self.drive(0, 180)
            elif intensity > LIGHTGREY:
                self.drive(30, 90)
            elif intensity > DARKGREY:
                self.drive(40, 0)
            elif intensity > BLACK:
                self.drive(30, -90)
            else:
                self.drive(0, -180)
                
            wait(1)
            
            if intensity > 10:
                watch_online.reset()
            elif (watch_online.time() > max_runtime or watch_online.time() > max_duration):
                print('exceeded.')
                break
        self.stop()
        
    def turn_left(self):
        print('turn left...')
        
        intensity = self.sensor.reflection()
        
        WHITE = 20
        LIGHTGREY = 14
        DARKGREY = 10
        BLACK = 7
        
        while intensity < WHITE:
            self.drive(0, -180)
            wait(1)
            
            if False:
                break
            
        self.stop()
```

closed 2021-11-25 by MarshToyokan

laurensvalk, 2021-11-25:
```python
        
        while intensity < WHITE:
            self.drive(0, -180)
            wait(1)
            
            if False:
                break
```
This loop never ends because `intensity` isn't updated inside the loop.

Try adding this inside the loop:

```python
intensity = self.sensor.reflection()
```

It's great to hear that Pybricks and Move Hub are being used for teaching by the way. May I ask what kind of class this is used for?

MarshToyokan, 2021-11-25:
Thank you very much! It worked!

I am a first-year university student in Japan and Pybricks and Move Hub are used in a lecture for programming!