pybricks/support #551: Apparent delay between some motor commands

laurensvalk, 2021-12-07:
----

**This issue was converted from a discussion in https://github.com/pybricks/support/discussions/550**

<div type='discussions-op-text'>

<sup>Originally posted by **@eadunes** December  6, 2021</sup>
Hi
currently working on a GBC module with 4x motors and 1x technic hub. I have a weird behavior.

```python
while True:
    hub.light.on(Color.GREEN)
    
    motorA.run_angle(600, -2640, wait=False)
    motorC.run_angle(300, 430, then=Stop.BRAKE)
    motorB.run_angle(900, -3600)
 
    motorC.run_angle(300, -430, wait=False)
    motorA.run_angle(600, 2640)
    motorD.run_angle(900, 3600)
    
    motorA.run_angle(600, -2640, wait=False)
    motorC.run_angle(300, 430, then=Stop.BRAKE)
    motorB.run_angle(900, 3600)
   
    motorC.run_angle(300, -430, wait=False)
    motorA.run_angle(600, 2640)
    motorD.run_angle(900, -3600)
```

after motorA.run_angle(600, 2640) in the 2nd and 4th paragraph is a pause of around 25 secs. I cant find out why.
any tips are appreciated.

Daniel

----

updated to the beta firmware . its way better but the pause is still there. it varies from 2-12 secs. without any seen pattern why. In general the longer the module runs the longer the pause. at the beginning its between 2-5 sec. later mostly between 7-9 sec. with spikes above and below.

----

```python
# Container Transporter
from pybricks.pupdevices import Motor
from pybricks.hubs import TechnicHub
from pybricks.parameters import Port, Color, Stop
from pybricks.tools import wait
# Initialize
hub = TechnicHub()
motorA = Motor(Port.A)
motorB = Motor(Port.B)
motorC = Motor(Port.C)
motorD = Motor(Port.D)
# Start Sequence
hub.light.on(Color.RED)
motorA.run_until_stalled(speed=800, duty_limit=20)
motorA.reset_angle(0)
motorC.run_until_stalled(speed=600, duty_limit=22)
motorC.reset_angle(0)
motorB.run_angle(800, 3600)
motorC.run_angle(400, -435)
motorD.run_angle(800, -3600)
# Sequence
while True:
    hub.light.on(Color.GREEN)   
    motorA.run_angle(600, -2640, wait=False)
    motorC.run_angle(300, 440, then=Stop.HOLD)
    motorB.run_angle(900, -3600)
    motorC.run_angle(300, -440, then=Stop.HOLD, wait=False)
    motorA.run_angle(600, 2640)
    motorD.run_angle(900, 3600)   
    motorA.run_angle(600, -2640, wait=False)
    motorC.run_angle(300, 440, then=Stop.HOLD)
    motorB.run_angle(900, 3600)   
    motorC.run_angle(300, -440, then=Stop.HOLD, wait=False)
    motorA.run_angle(600, 2640)
    motorD.run_angle(900, -3600)
```

here is the code. and think I found the reason for the behavoir. in the Start Sequence I use the run_until_stalled command to set angle to Zero with duty_limit. it was 30 for motorA. which had the pause issue. then I lowered it to 25. and the pause got shorter. now set to 20 and the next motor starts immediately.
So now the question. is the duty limit for the motors global ? I thought it was just for the command it's written in.

I will record a 10 mins video tomorrow with the new settings.

Daniel

</div>

laurensvalk, 2021-12-07:
Thanks for reporting! To find out why this is happening, it would be really useful if you want to log some data for us. It works like this:

```python

eadunes, 2021-12-07:
duty_limit 30

PB_OF:servo.txt
48,330917022,-2591,0,3,1140,-2605,104,-3819,33368
98,330967022,-2582,61,3,828,-2596,240,-25819,47273
148,331017024,-2571,137,3,2368,-2582,362,190,61178
198,331067034,-2553,228,3,3832,-2560,481,24199,75082
248,331117022,-2529,381,3,3359,-2533,600,18280,68742
298,331167022,-2502,472,3,3984,-2502,598,34484,68742
348,331217022,-2471,533,3,3609,-2472,601,24769,68742
398,331267022,-2438,594,3,3540,-2442,596,22982,68742
448,331317022,-2403,671,3,2076,-2412,605,-14949,68742
498,331367022,-2374,671,3,2540,-2382,601,-2937,68742
548,331417023,-2345,610,3,2851,-2352,599,5117,68742
598,331467022,-2319,564,3,3704,-2322,596,27228,68742
648,331517023,-2292,533,3,2633,-2291,616,-511,68742
698,331567023,-2257,564,3,2565,-2261,607,-2292,68742
748,331617023,-2222,671,3,2257,-2231,601,-10250,68742
798,331667022,-2193,671,3,3338,-2201,589,17750,68742
848,331717022,-2168,579,3,3882,-2171,592,31828,68742
898,331767022,-2141,533,3,3271,-2141,606,16011,68742
948,331817022,-2111,533,3,3282,-2111,606,16281,68742
998,331867022,-2079,594,3,2518,-2080,612,-3497,68742
1048,331917023,-2044,655,3,2444,-2050,603,-5407,68742
1098,331967022,-2013,671,3,3448,-2020,588,20596,68742
1148,332017023,-1984,625,3,2755,-1990,599,2629,68742
1198,332067022,-1958,579,3,3607,-1961,596,24707,68742
1248,332117023,-1930,533,3,3229,-1930,605,14911,68742
1298,332167022,-1896,579,3,2926,-1900,601,7076,68742
1348,332217022,-1861,655,3,2155,-1870,601,-12903,68742
1398,332267022,-1833,671,3,3390,-1841,589,19097,68742
1448,332317022,-1808,579,3,3085,-1811,603,11184,68742
1498,332367022,-1782,518,3,3093,-1780,611,11388,68742
1548,332417022,-1755,518,3,3415,-1750,613,19724,68742
1598,332467022,-1724,549,3,3892,-1719,605,32087,68742
1648,332517022,-1692,610,3,3286,-1688,609,16390,68742
1698,332567021,-1660,640,3,3450,-1658,603,20630,68742
1748,332617022,-1631,625,3,2764,-1627,614,2864,68742
1798,332667021,-1604,594,3,2696,-1597,621,1110,68742
1848,332717022,-1577,549,3,3792,-1566,613,29503,68742
1898,332767021,-1545,549,3,3498,-1535,613,21898,68742
1948,332817022,-1511,640,3,2885,-1505,613,6000,68742
1998,332867021,-1480,671,3,3657,-1475,601,26000,68742
2048,332917022,-1454,610,3,4352,-1444,600,44000,68742
2098,332967022,-1427,549,3,4120,-1413,609,38000,68742
2148,333017023,-1398,533,3,4197,-1383,610,40000,68742
2198,333067022,-1364,579,3,3966,-1352,605,34000,68742
2248,333117021,-1330,671,3,4043,-1322,596,36000,68742
2298,333167022,-1293,716,3,3116,-1292,594,12000,68742
2348,333217022,-1265,686,3,3888,-1262,588,32000,68742
2398,333267022,-1236,610,3,3348,-1232,597,18000,68742
2448,333317023,-1206,564,3,2730,-1202,605,2000,68742
2498,333367022,-1176,579,3,4274,-1172,585,42000,68742
2548,333417022,-1142,655,3,3734,-1143,584,28000,68742
2598,333467021,-1112,671,3,3811,-1113,583,30000,68742
2648,333517023,-1085,610,3,3116,-1083,598,12000,68742
2698,333567022,-1061,533,3,3116,-1053,610,12000,68742
2748,333617022,-1033,518,3,3502,-1023,609,22000,68742
2798,333667021,-1002,549,3,2962,-992,614,8000,68742
2848,333717022,-968,625,3,3194,-962,603,14000,68742
2898,333767021,-937,655,3,3734,-932,594,28000,68742
2948,333817022,-908,625,3,3348,-902,601,18000,68742
2998,333867021,-881,579,3,3116,-872,610,12000,68742
3048,333917022,-856,533,3,4197,-841,606,40000,68742
3098,333967021,-822,549,3,3966,-811,601,34000,68742
3148,334017022,-785,655,3,2885,-781,601,6000,68742
3198,334067021,-755,701,3,2653,-751,604,0,68742
3248,334117022,-729,610,3,3039,-721,607,10000,68742
3298,334167021,-703,533,3,3734,-691,606,28000,68742
3348,334217022,-675,518,3,3271,-660,616,16000,68742
[... 312 lines cut ...]
8898,8897093,-31,0,3,68100,-20,0,-2,-5,44000,14100,10000,-4465
8948,8947094,-31,0,3,68250,-20,0,-2,-5,44000,14250,10000,134359868
8998,8997093,-31,0,3,68400,-20,0,-2,-5,44000,14400,10000,-4465
9048,9047094,-31,0,3,68550,-20,0,-2,-5,44000,14550,10000,134359868
9098,9097093,-31,0,3,68700,-20,0,-2,-5,44000,14700,10000,-4465
9148,9147094,-31,0,3,68850,-20,0,-2,-5,44000,14850,10000,134359868
9198,9197093,-31,0,3,69000,-20,0,-2,-5,44000,15000,10000,536923604
9248,9247094,-31,0,3,69150,-20,0,-2,-5,44000,15150,10000,134359868
9298,9297093,-31,0,3,69300,-20,0,-2,-5,44000,15300,10000,-4465
9348,9347094,-31,0,3,69450,-20,0,-2,-5,44000,15450,10000,536923552
9398,9397093,-31,0,3,69600,-20,0,-2,-5,44000,15600,10000,-4465
9448,9447094,-31,0,3,69750,-20,0,-2,-5,44000,15750,10000,134359868
9498,9497093,-31,0,3,69900,-20,0,-2,-5,44000,15900,10000,-4465
9548,9547094,-31,0,3,70050,-20,0,-2,-5,44000,16050,10000,134359868
9598,9597093,-31,0,3,70200,-20,0,-2,-5,44000,16200,10000,-4465
9648,9647094,-31,0,3,70350,-20,0,-2,-5,44000,16350,10000,134359868
9698,9697093,-31,0,3,70500,-20,0,-2,-5,44000,16500,10000,-4465
9748,9747094,-31,0,3,70650,-20,0,-2,-5,44000,16650,10000,134359868
9798,9797093,-31,0,3,70800,-20,0,-2,-5,44000,16800,10000,-4465
9848,9847094,-31,0,3,70950,-20,0,-2,-5,44000,16950,10000,134359868
9898,9897093,-31,0,3,71100,-20,0,-2,-5,44000,17100,10000,-4465
9948,9947094,-31,0,3,71250,-20,0,-2,-5,44000,17250,10000,134359868
9998,9997093,-31,0,3,71400,-20,0,-2,-5,44000,17400,10000,-4465
PB_EOF
Done

eadunes, 2021-12-07:
duty_limit 25

PB_OF:servo.txt
48,509650078,-2586,0,3,1287,-2599,104,-15,33368
98,509700114,-2576,61,3,2133,-2590,223,7985,47273
148,509750049,-2565,137,3,2129,-2576,365,-6015,61178
198,509800046,-2546,228,3,3515,-2554,483,15997,75082
248,509850049,-2521,381,3,2733,-2527,604,2073,68742
298,509900047,-2495,488,3,2739,-2496,612,2229,68742
348,509950049,-2465,533,3,3366,-2466,604,18466,68742
398,510000047,-2432,579,3,3220,-2436,600,14672,68742
448,510050050,-2399,655,3,2605,-2406,602,-1238,68742
498,510100047,-2368,655,3,2530,-2376,601,-3187,68742
548,510150050,-2339,594,3,3460,-2346,591,20894,68742
598,510200048,-2312,579,3,2615,-2316,608,-1000,68742
648,510250050,-2284,564,3,3317,-2286,603,17204,68742
698,510300047,-2249,579,3,2474,-2255,604,-4640,68742
748,510350078,-2215,671,3,2939,-2225,590,7396,68742
798,510400048,-2187,671,3,2244,-2196,603,-10595,68742
848,510450079,-2162,579,3,3175,-2165,601,13510,68742
898,510500046,-2139,518,3,4111,-2135,603,37774,68742
948,510550049,-2110,518,3,3432,-2104,614,20167,68742
998,510600047,-2077,579,3,2982,-2073,614,8524,68742
1048,510650050,-2040,655,3,2913,-2043,601,6734,68742
1098,510700048,-2007,701,3,2761,-2013,597,2782,68742
1148,510750051,-1977,671,3,3301,-1984,590,16788,68742
1198,510800047,-1951,594,3,2608,-1954,607,-1167,68742
1248,510850049,-1924,533,3,3851,-1923,597,31022,68742
1298,510900046,-1890,564,3,2390,-1893,608,-6813,68742
1348,510950049,-1856,655,3,3009,-1863,592,9211,68742
1398,511000046,-1827,655,3,2932,-1833,595,7214,68742
1448,511050049,-1803,579,3,3399,-1804,601,19310,68742
1498,511100046,-1776,518,3,3406,-1773,607,19508,68742
1548,511150049,-1747,533,3,3109,-1743,613,11805,68742
1598,511200047,-1717,564,3,3198,-1712,612,14126,68742
1648,511250049,-1684,610,3,3517,-1682,602,22387,68742
1698,511300046,-1653,640,3,3603,-1651,599,24597,68742
1748,511350049,-1624,610,3,2684,-1621,613,804,68742
1798,511400045,-1597,594,3,3696,-1591,606,27023,68742
1848,511450049,-1570,549,3,4019,-1560,608,35386,68742
1898,511500046,-1537,549,3,3801,-1529,605,29740,68742
1948,511550049,-1503,640,3,3038,-1499,607,9962,68742
1998,511600046,-1474,671,3,3348,-1469,605,17995,68742
2048,511650050,-1447,594,3,3116,-1438,614,11995,68742
2098,511700046,-1421,533,3,4120,-1408,609,37995,68742
2148,511750049,-1393,518,3,3579,-1377,620,23995,68742
2198,511800045,-1358,579,3,3657,-1346,609,25995,68742
2248,511850049,-1321,686,3,3734,-1316,594,27995,68742
2298,511900045,-1288,716,3,3348,-1286,593,18003,68742
2348,511950049,-1256,671,3,3657,-1256,585,26000,68742
2398,512000046,-1230,625,3,3502,-1226,595,22000,68742
2448,512050049,-1200,564,3,2576,-1196,607,-2000,68742
2498,512100046,-1166,579,3,3194,-1166,591,14000,68742
2548,512150049,-1132,655,3,3425,-1137,580,20000,68742
2598,512200046,-1104,655,3,3425,-1107,584,20000,68742
2648,512250050,-1078,579,3,3657,-1078,589,26000,68742
2698,512300047,-1053,518,3,4043,-1048,594,36000,68742
2748,512350049,-1025,518,3,3271,-1018,608,16000,68742
2798,512400046,-996,549,3,4274,-987,597,42000,68742
2848,512450050,-964,579,3,3116,-957,608,12000,68742
2898,512500046,-933,610,3,2808,-926,610,4000,68742
2948,512550050,-904,610,3,2885,-896,611,6000,68742
2998,512600047,-880,594,3,3425,-866,616,20000,68742
3048,512650051,-849,549,3,4352,-835,602,44000,68742
3098,512700046,-817,579,3,4120,-805,601,38000,68742
3148,512750102,-778,686,3,2808,-775,600,4000,68742
3198,512800048,-752,686,3,4043,-745,592,36000,68742
3248,512850049,-725,594,3,3966,-715,599,34000,68742
3298,512900046,-697,549,3,3116,-684,614,12000,68742
3348,512950049,-667,549,3,3039,-654,615,10000,68742
[... 312 lines cut ...]
8898,8897123,-25,0,3,68100,-14,0,3,-5,44000,14100,10000,35
8948,8947127,-25,0,3,68250,-14,0,3,-5,44000,14250,10000,35
8998,8997124,-25,0,3,68400,-14,0,3,-5,44000,14400,10000,35
9048,9047127,-25,0,3,68550,-14,0,3,-5,44000,14550,10000,35
9098,9097117,-25,0,3,68700,-14,0,3,-5,44000,14700,10000,35
9148,9147120,-25,0,3,68850,-14,0,3,-5,44000,14850,10000,35
9198,9197118,-25,0,3,69000,-14,0,3,-5,44000,15000,10000,35
9248,9247121,-25,0,3,69150,-14,0,3,-5,44000,15150,10000,35
9298,9297117,-25,0,3,69300,-14,0,3,-5,44000,15300,10000,35
9348,9347120,-25,0,3,69450,-14,0,3,-5,44000,15450,10000,35
9398,9397118,-25,0,3,69600,-14,0,3,-5,44000,15600,10000,35
9448,9447120,-25,0,3,69750,-14,0,3,-5,44000,15750,10000,35
9498,9497117,-25,0,3,69900,-14,0,3,-5,44000,15900,10000,35
9548,9547123,-25,0,3,70050,-14,0,3,-5,44000,16050,10000,35
9598,9597123,-25,0,3,70200,-14,0,3,-5,44000,16200,10000,35
9648,9647127,-25,0,3,70350,-14,0,3,-5,44000,16350,10000,35
9698,9697124,-25,0,3,70500,-14,0,3,-5,44000,16500,10000,35
9748,9747127,-25,0,3,70650,-14,0,3,-5,44000,16650,10000,0
9798,9797124,-25,0,3,70800,-14,0,3,-5,44000,16800,10000,134359596
9848,9847128,-25,0,3,70950,-14,0,3,-5,44000,16950,10000,9
9898,9897124,-25,0,3,71100,-14,0,3,-5,44000,17100,10000,35
9948,9947127,-25,0,3,71250,-14,0,3,-5,44000,17250,10000,35
9998,9997123,-25,0,3,71400,-14,0,3,-5,44000,17400,10000,35
PB_EOF
Done

eadunes, 2021-12-07:
duty_limit 20

PB_OF:servo.txt
49,653969021,-2587,0,3,616,-2604,114,-17744,33702
99,654019023,-2579,30,3,2002,-2595,232,4256,47607
149,654069022,-2571,106,3,2771,-2580,368,10286,61511
199,654119023,-2546,228,3,3927,-2558,479,26328,75416
249,654169022,-2522,411,3,2590,-2531,604,-1639,68742
299,654219023,-2496,503,3,3676,-2500,598,26499,68742
349,654269063,-2470,518,3,3455,-2470,609,20772,68742
399,654319023,-2439,533,3,3391,-2439,608,19102,68742
449,654369021,-2409,594,3,3170,-2409,611,13384,68742
499,654419023,-2373,655,3,2328,-2378,610,-8424,68742
549,654469022,-2345,655,3,2489,-2348,612,-4250,68742
599,654519022,-2314,625,3,3731,-2318,594,27921,68742
649,654569022,-2288,594,3,3198,-2288,609,14116,68742
699,654619022,-2256,579,3,2590,-2257,613,-1635,68742
749,654669022,-2219,625,3,3289,-2227,590,16476,68742
799,654719023,-2185,701,3,2749,-2197,589,2476,68742
849,654769022,-2159,655,3,3058,-2168,593,10476,68742
899,654819023,-2133,549,3,2983,-2138,602,8548,68742
949,654869021,-2108,518,3,3917,-2108,600,32737,68742
999,654919022,-2076,549,3,3077,-2077,607,10968,68742
1049,654969022,-2042,610,3,2542,-2047,606,-2879,68742
1099,655019022,-2010,655,3,2930,-2017,597,7166,68742
1149,655069022,-1981,640,3,3394,-1987,593,19202,68742
1199,655119022,-1953,579,3,2857,-1957,604,5292,68742
1249,655169022,-1928,564,3,2710,-1927,616,1466,68742
1299,655219022,-1898,533,3,3107,-1896,611,11748,68742
1349,655269022,-1864,579,3,3655,-1866,596,25961,68742
1399,655319022,-1829,671,3,3504,-1836,588,22036,68742
1449,655369022,-1802,655,3,3737,-1806,591,28069,68742
1499,655419022,-1774,579,3,2660,-1776,609,189,68742
1549,655469021,-1748,549,3,3055,-1745,612,10399,68742
1599,655519022,-1719,533,3,4070,-1715,601,36696,68742
1649,655569021,-1686,579,3,3154,-1684,607,12975,68742
1699,655619022,-1655,625,3,2930,-1654,608,7173,68742
1749,655669021,-1624,640,3,3631,-1624,597,25329,68742
1799,655719022,-1595,610,3,3406,-1594,602,19494,68742
1849,655769021,-1570,579,3,3337,-1563,613,17719,68742
1899,655819022,-1541,533,3,3194,-1532,617,14000,68742
1949,655869021,-1510,549,3,3888,-1501,606,32000,68742
1999,655919022,-1476,625,3,2885,-1471,611,6000,68742
2049,655969021,-1447,655,3,4043,-1441,598,36000,68742
2099,656019022,-1420,579,3,3888,-1410,606,32000,68742
2149,656069022,-1393,549,3,3271,-1380,620,16000,68742
2199,656119022,-1362,564,3,4197,-1349,606,40000,68742
2249,656169022,-1326,610,3,2885,-1318,611,6000,68742
2299,656219024,-1293,686,3,3039,-1288,603,10000,68742
2349,656269022,-1261,686,3,3194,-1258,597,14000,68742
2399,656319022,-1234,640,3,3888,-1228,594,32000,68742
2449,656369045,-1206,594,3,3502,-1198,603,22000,68742
2499,656419022,-1175,549,3,3348,-1168,603,18000,68742
2549,656469021,-1137,640,3,3734,-1138,582,28000,68742
2599,656519022,-1101,747,3,2113,-1109,591,-14000,68742
2649,656569021,-1075,686,3,2653,-1079,592,0,68742
2699,656619022,-1049,564,3,2730,-1050,599,2000,68742
2749,656669043,-1025,503,3,3348,-1020,603,18000,68742
2799,656719022,-996,518,3,3348,-989,605,18000,68742
2849,656769021,-965,564,3,3425,-959,602,20000,68742
2899,656819022,-935,610,3,2962,-929,608,8000,68742
2949,656869021,-905,610,3,2808,-899,610,4000,68742
2999,656919022,-876,594,3,3271,-869,606,16000,68742
3049,656969021,-852,564,3,3348,-838,617,18000,68742
3099,657019022,-821,533,3,3580,-807,612,24000,68742
3149,657069021,-785,594,3,3348,-777,603,18000,68742
3199,657119022,-750,701,3,3888,-747,586,32000,68742
3249,657169021,-723,686,3,2962,-717,604,8000,68742
3299,657219022,-695,579,3,3811,-687,597,30000,68742
3349,657269021,-669,533,3,3502,-657,609,22000,68742
[... 246 lines cut ...]
8899,8898129,-24,0,3,44100,-16,0,-2,1,32000,14100,-2000,8321
8949,8948130,-24,0,3,46250,-16,0,-2,0,32000,14250,0,8321
8999,8998094,-23,0,3,40400,-16,0,-2,1,28000,14400,-2000,536924064
9049,9048096,-23,0,3,42550,-16,0,-2,0,28000,14550,0,536923964
9099,9098094,-24,0,3,46700,-16,0,-2,0,32000,14700,0,536923964
9149,9148093,-24,0,3,46850,-16,0,-2,0,32000,14850,0,536923964
9199,9198092,-24,0,3,47000,-16,0,-2,0,32000,15000,0,536923964
9249,9248093,-24,0,3,47150,-16,0,-2,0,32000,15150,0,536923964
9299,9298094,-24,0,3,47300,-16,0,-2,0,32000,15300,0,536923964
9349,9348093,-24,0,3,47450,-16,0,-2,0,32000,15450,0,536923964
9399,9398096,-24,0,3,47600,-16,0,-2,0,32000,15600,0,536923964
9449,9448095,-24,0,3,47750,-16,0,-2,0,32000,15750,0,536923964
9499,9498118,-24,0,3,47900,-16,0,-2,0,32000,15900,0,536923964
9549,9548094,-23,0,3,44050,-16,0,-2,0,28000,16050,0,536923964
9599,9598094,-24,0,3,48200,-16,0,-2,0,32000,16200,0,536923964
9649,9648093,-24,0,3,48350,-16,0,-2,0,32000,16350,0,536923964
9699,9698096,-23,0,3,44500,-16,0,-2,0,28000,16500,0,536923964
9749,9748094,-23,0,3,44650,-16,0,-2,0,28000,16650,0,536923964
9799,9798107,-23,0,3,44800,-16,0,-2,0,28000,16800,0,536923964
9849,9848131,-24,0,3,48950,-16,0,-2,0,32000,16950,0,8321
9899,9898125,-24,0,3,49100,-16,0,-2,0,32000,17100,0,8321
9949,9948094,-24,0,3,47250,-16,0,-2,1,32000,17250,-2000,536924064
9999,9998094,-24,0,3,49400,-16,0,-2,0,32000,17400,0,536923964
PB_EOF
Done

eadunes, 2021-12-07:
Hardware: PU Technic hub
                  Motor A = 88014
                  Motor B = 88008
                  Motor C = 88014
                  Motor D = 88008

eadunes, 2021-12-08:
here is the video
[https://youtu.be/OwHmByGHJwk]
config is as posted above

laurensvalk, 2021-12-08:
Thanks a lot for taking the time to make the logs and the videos!

I have processed your data into plots that show the motor behavior over time.

It looks like the issue is the same in all cases. I think `duty_limit` is not the issue itself: a higher limit just puts the motor closer to your mechanical endstop. And since you are going back to this mechanical stop later in your code, the motor has to work a bit harder when it is closer to the mechanical stop.

Now on to the main issue:

----

Have a look at the first two graphs in the "control plot. In both cases, look at the blue lines.

You can see that the motor goes from -2640 to 0. Well, almost! In the second plot you can see the reported error, which is how many degrees it is away from its target.

So it's still about 10 degrees short of its final target.

Now in the torque plot (second to last), you can see that the I term (that's the I from PID) starts to rise as expected. This is normally enough to give it that final push to reach the target.

But it is rising very slowly.

Here are two things you can do:

1. Either increase the `ki` gain, like this.
```python
# Read the default settings
default_settings = motorA.control.pid()
default_ki = default_settings[1]
# Increase the integral control by a factor of 4 (you can choose any factor that works for you)
motorA.control.pid(ki=default_ki * 4)
```

2. Or, move a little bit away from the end stop

   After you do `run_until_stalled`, move away a little bit, like this: ``motorA.run_angle(600, -20)``

   Then for the remainder if your code, you are not pushing against the end point every time. This avoids the problem,
   and it should look a bit smoother as well. Your batteries will also last a bit longer since the motor doesn't need to work
   so hard.

Option 2 might make the most sense in your application.

But we can probably do option 1 in the next release of Pybricks so it will also be solved for others.

Thanks again for reporting this, and let me know if or or both options work.

laurensvalk, 2021-12-08:
After taking a closer look, I can confirm that the default `ki` value for the Control+ XL motor is much too low. When it is stuck at 10 degrees away from its target, it takes about `260000 / 10 / 600 = 40` seconds to reach its maximum torque, or 20 seconds to reach half of it. That's close to what you were seeing.

eadunes, 2021-12-09:
I took option 2 cos it was easier for me to implement.
[image]

here the video of it running. 
https://youtu.be/ePiUr0KToXA
if U look closer there is small pause around 30 secs before the end when the green bin got emptied. 
already fixed this by setting motorA -45 degrees instead of -40 in the start sequence.

thanks for the help.

br Daniel

closed 2022-01-18 by laurensvalk