pybricks/support #552: [Bug] Black formatting version

laurensvalk, 2021-12-07:
**Describe the bug**
CI builds have started failing on the format task. 

It appears that new builds use

```
Collecting black
  Downloading black-21.12b0-py3-none-any.whl 
```

While previous builds used 

```
Collecting black
  Downloading black-21.11b1-py3-none-any.whl (155 kB)
```

Maybe the right version can be locked in our Poetry config file.

dlech, 2021-12-07:
There is already a poetry.lock file that does this. It must have been updated. So we should just run the new version of black on the code and commit it.

dlech, 2021-12-07:
Which repository are you referring to?

laurensvalk, 2021-12-07:
For example, this branch is failing, even though 4 commits down it was passing:
https://github.com/pybricks/pybricks-micropython/commits/broadcast2

In between those, the script that causes it to fail hasn't changed, and neither has the lock file.

```
Run ./tools/codeformat.py
  ./tools/codeformat.py
  git diff --exit-code
  shell: /usr/bin/bash -e {0}
  env:
    pythonLocation: /opt/hostedtoolcache/Python/3.6.15/x64
    LD_LIBRARY_PATH: /opt/hostedtoolcache/Python/3.6.15/x64/lib
diff --git a/tests/pup/benchmark/pystone.py b/tests/pup/benchmark/pystone.py
index 7dc2792..2dfe934 100644
--- a/tests/pup/benchmark/pystone.py
+++ b/tests/pup/benchmark/pystone.py
@@ -27,7 +27,6 @@ try:
     def clock():
         return watch.time() / 1000
 
-
 except ImportError:
     from utime import ticks_ms
 
Error: Process completed with exit code 1.
```

dlech, 2021-12-07:
Ah, we are just installing black in CI without using poetry (probably because it is much faster than installing all of the other deps).

https://github.com/pybricks/pybricks-micropython/blob/dc311ca3f3762d2e8ff75f66fffbd521c542be49/.github/workflows/format.yml#L40

The easy fix is probably to just update poetry to the latest version of black.

closed 2021-12-07 by dlech