pybricks/support #554: [Question] blutooth connection problem at macOS

ErikLundJensen, 2021-12-12:
**Question**
I try to run pybricks at MacBook Pro 2019 macOs 12.0.1, however, pybricks can't pair with LEGO Technic Hub.
In chrome://bluetooth-internals I can connect and get GATT Connected so it does not seem to be a Chrome nor Mac issue to form pair.

There are no errors from pybricks - just nothing happens when trying to connect -- keeps on waiting for pairing.

I'm new to pybricks, but should I expect bluetooth problems when using macOS with pybricks?

**Context**
Connecting to Technic Hub from macOs 12.0.1

dlech, 2021-12-12:
Did you flash the Pybricks firmware first?

ErikLundJensen, 2021-12-13:
The flash dialog also hangs without getting a connection.

laurensvalk, 2021-12-13:
Did you put the hub in update mode by holding the hub button until it blinks purple?

Instructions and a video can be found here: https://pybricks.com/install/technic-boost-city/

ErikLundJensen, 2021-12-15:
I tried that, the purple continues to flash and the hub is not shown in UI. I'll try with another Technic hub later this month.

dlech, 2021-12-15:
Are you clicking the firmware flash button or the Bluetooth button in Pybricks Code?

ErikLundJensen, 2021-12-16:
the firmware flash button.

dlech, 2021-12-16:
It sounds like you are doing it right. We have some other troubleshooting tips in #270 but I'm not sure what could be wrong in this case.

injeniero, 2022-01-12:
@dlech @ErikLundJensen I had to enable chrome://flags/#enable-experimental-web-platform-features for it to work. I'm running on MacBook Pro 2019 macOs 12.1.0 with chrome Version 97.0.4692.71 after enabling the flag and restarting the browser, I was able to connect.

dlech, 2022-01-12:
> I had to enable chrome://flags/#enable-experimental-web-platform-features for it to work.

This should only be required on Linux. If you turn it back off does it still work?

injeniero, 2022-01-12:
I just tried disabling it and it continues to work. I can only guess by setting this it forced me to restart chrome and that made it work.

dlech, 2023-03-21:
Closing since this issue is a few years old and lots has changed since then. Any future visitors experience connection issues on macOS: please log Bluetooth packets and share them in #938 so we can try to figure out what is going on.

closed 2023-03-21 by dlech