pybricks/support #555: Please see #270 for troubleshooting tips and open a new issue with logs/packet capture.

VegaMansk, 2021-12-13:
Please see #270 for troubleshooting tips and open a new issue with logs/packet capture.

_Originally posted by @dlech in https://github.com/pybricks/support/issues/167#issuecomment-991932731_

VegaMansk, 2021-12-13:
Error on code.pybricks.com screen . After connection thrue bluetooth. I get the message unexpected error while trying to connect: Connection failed for unknown reason.

I use: Windows 10 Education: Versie: 10.0.19044 Build 19044. Chrome browser Versie 96.04.4664.93 (64 bits). The update of My Inventor (51515) with pybricks (with USB) went ok. Do you have advise for me?

[Bluetoothlog.zip](https://github.com/pybricks/support/files/7704565/Bluetoothlog.zip)

dlech, 2021-12-13:
Thanks for making the logs. It looks like the connection is successful in the logs, so I'm afraid I don't know why Windows is giving an error. Did you try rebooting the computer? If there are any LEGO or Pybricks hubs listed in *Bluetooth &  other devices* in the WIndows settings, try removing them.

[image]

MartinGarciaMatienzo, 2021-12-22:
I have the same issue with my technic hubs, I wanted to connect 2 hubs from the Liebherr set, but my laptop pybricks doesn't connect to the hubs but it registers the hubs. Is there any way I can connect them properly or some solutions..? 
It says "Vinculado", meaning its connected, but the time I wanna pair them it just take a while and sends the unexpected error while trying to conect.
Thanks for reading.
[image]

MartinGarciaMatienzo, 2021-12-22:
Even when I try to connect directly by the PC's Bluetooth it shows "LEGO Boothloader" as in pybricks, but sometimes when the LED is white; it says "Technic Hub".
[image]

MartinGarciaMatienzo, 2021-12-22:
Now just says: "GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`."
[image]

dlech, 2021-12-22:
> Even when I try to connect directly by the PC's Bluetooth it shows "LEGO Boothloader"

Do not pair the hub using the OS settings menu. If you have already done this, remove the device from the settings menu.

Also see #270 for troubleshooting tips. If you can log Bluetooth packets, please upload them here and we will have a look.

MartinGarciaMatienzo, 2022-01-01:
> Do not pair the hub using the OS settings menu. If you have already done this, remove the device from the settings menu.
>
> Also see #270 for troubleshooting tips. If you can log Bluetooth packets, [...]

I didn´t, I don't know what the problem is. The connection, upload from the firmware and the program ran well in an ASUS computer, I'm trying to connect with an HP but doesn´t. Both use Windows 10