pybricks/support #556: ``` GATT operation not permitted. ```

Valter1966, 2021-12-16:
Bluetooth connection problem whit windows 10
Unexpected error while trying to connect: GATT operation not permitted.

dlech, 2021-12-16:
I assume this is happening on <https://code.pybricks.com>? Please see our troubleshooting guide at #270 and post logs/screenshots here if there is not a simple fix like rebooting your computer.

Valter1966, 2021-12-19:
Hi,
Yes, this is happening on https://code.pybricks.com when i try to connect the brick whit the PC whit Bluetooth.
My TechnicHub is working (with other PCs it is OK) and it is also recognized by my PC (see screenshot).

[image]

But when I try to pair it the following error comes out

[image]

Thanks in advance for your support
:-)

dlech, 2022-06-03:
We cannot solve the issue without more detailed logs, like `chome://device-logs` or a Bluetooth packet capture with wireshark.