pybricks/support #559: [Feature] Make low battery indicator more visible

laurensvalk, 2021-12-21:
**Is your feature request related to a problem? Please describe.**

Our low-battery indicator currently looks like this:

[image]

But it can be easy to miss and users may not know what it means.

**Describe the solution you'd like**

We should keep the light, but we could also have one of the following to make it clearer:

- Raise an appropriate exception. Perhaps an `OSError` or the `SystemExit` like we do for the button press, but with a different message.
- Battery status in Pybricks Code (see #173)

laurensvalk, 2021-12-21:
I like the exception because it does not require much change in Pybricks Code, and keeps it generic with Pybricksdev, etc.

dlech, 2021-12-21:
The low battery warning just means that the battery won't last much longer, but you can still keep using your hub until it shuts itself off. If we raise an exception or prevent the user program from starting, then it effectively reduces the amount of time you can use your hub. With disposable batteries, it means you would have to throw them out even though there is still some useful life left in them.

closed 2021-12-28 by dlech