pybricks/support #561: [Bug] Motor can sometimes raise EBUSY when starting new script.

laurensvalk, 2021-12-23:
**Describe the bug**
`OSError/EBUSY` should only be raised if initializing a motor for the second time in the same program.

But it is also raised during our tests. Perhaps it does not show in some individual tests but can happen when some sequence of programs is run, indicating a bad reset of the motor state.

This does not affect V3.1.

**To reproduce**
Run `tests/pup/run_all.py`.

**Expected behavior**
All relevant tests to pass.

**Screenshots**
```
Now running: ./hub/status_light.py
 83%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████▌                       | 200/241 [00:00<00:00, 672B/s]Color: BROWN
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 241/241 [00:00<00:00, 603B/s]
Color: YELLOW
Color: GREEN
Color: CYAN
Color: BLACK
Color: NONE
Color: MAGENTA
Color: RED
Color: BLUE
Color: VIOLET
Color: GRAY
Color: ORANGE
Color: WHITE
...done
Now running: ./hub/status_light_animation.py
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 285/285 [00:00<00:00, 633B/s]
Now running: ./motors/busy.py
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 312/312 [00:00<00:00, 568B/s]
Traceback (most recent call last):
  File "./motors/busy.py", line 28, in <module>
NameError: name 'exception' isn't defined
Saving bad output to ./motors/busy.out
Now running: ./motors/dcmotor.py
 97%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████▋    | 400/413 [00:00<00:00, 655B/s]Traceback (most recent call last):
  File "./motors/dcmotor.py", line 15, in <module>
OSError: [Errno 19] ENODEV: 

100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 413/413 [00:00<00:00, 580B/s]
A sensor or motor is not connected to the specified port:
--> Check the cables to each motor and sensor.
--> Check the port settings in your script.
--> Check the line in your script that matches
    the line number given in the 'Traceback' above.

Saving bad output to ./motors/dcmotor.out
Now running: ./motors/gears.py
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 620/620 [00:00<00:00, 621B/s]
Traceback (most recent call last):
  File "./motors/gears.py", line 57, in <module>
OSError: [Errno 16] EBUSY: Device or resource busy
Saving bad output to ./motors/gears.out
Now running: ./motors/measurements.py
 95%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████▊      | 1.60k/1.68k [00:02<00:00, 667B/s]Traceback (most recent call last):
  File "./motors/measurements.py", line 16, in <module>
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 1.68k/1.68k [00:02<00:00, 654B/s]
OSError: [Errno 16] EBUSY: Device or resource busy
Saving bad output to ./motors/measurements.out
Now running: ./motors/trajectory.py
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 680/680 [00:01<00:00, 646B/s]
Now running: ./sensors/color_arguments.py
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 585/585 [00:00<00:00, 649B/s]
Now running: ./sensors/color_color.py
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 691/691 [00:01<00:00, 658B/s]
Now running: ./sensors/color_default.py
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 385/385 [00:00<00:00, 639B/s]
Now running: ./sensors/color_mode.py
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 498/498 [00:00<00:00, 663B/s]
Now running: ./sensors/color_reflection.py
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 417/417 [00:00<00:00, 762B/s]
Now running: ./sensors/ultrasonic_distance.py
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 583/583 [00:00<00:00, 834B/s]
Now running: ./sensors/ultrasonic_lights.py
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 377/377 [00:00<00:00, 754B/s]
Now running: ./sensors/ultrasonic_presence.py
100%|███████████████████████████████████████████████████████████████████████████
```

laurensvalk, 2022-01-03:
This is expected behavior, the tests just need to be updated accordingly. Specifically, initializing the motor more than once is not allowed:

```python
test = Motor(Port.A)
test2 = Motor(Port.A)  # Raises EBUSY
```

If we wanted to allow it, we should implement `__del__`, but I don't think this is worth the space. And initializing a motor just once is encouraged; that's why we raise the exception in the first place.

laurensvalk, 2022-01-03:
This is still allowed for `DCMotor`, though it shouldn't be. So we need to fix that as well.

dlech, 2022-01-03:
> If we wanted to allow it, we should implement `__del__`,

FWIW, this is not what `__del__` is for. It should have a `.close()` method instead.

laurensvalk, 2022-12-20:
The following test can't be fixed with the current firmware, so I'll delete it and archive it here.

```python
# SPDX-License-Identifier: MIT
# Copyright (c) 2020 The Pybricks Authors

"""
Hardware Module: 1

Description: Verifies gears argument for motors.
"""

from pybricks.pupdevices import Motor
from pybricks.parameters import Port
from pybricks.tools import wait

def get_gear_ratio(gear_arg):
    """Converts gear lists to gear ratio."""
    # Default ratio:
    if gear_arg is None:
        return 1.0

    # Ratio for single gear train.
    if type(gear_arg[0]) not in (list, tuple):
        return gear_arg[-1] / gear_arg[0]

    # Ratio for multiple gear trains.
    ratio = 1.0
    for train in gear_arg:
        ratio *= train[-1] / train[0]
    return ratio

# Gear arguments to test.
test_args = [
    None,
    [1, 3],
    (1, 3),
    [1, 2, 3],
    [1, 28],
    [12, 36],
    [36, 12],
    [[1, 24], [12, 36]],
    [(1, 24), (12, 36)],
    [(1, 24), [12, 36]],
    [[57, 24], [12, 123456789, 1]],
]

# Initialize device to known position.
motor = Motor(Port.A, gears=None)
motor.run_target(500, 170)
wait(1000)
motor.dc(0)
real_angle = motor.angle()

# Test expected result.
for gears in test_args:
    # Initialize motors with new settings.
    motor = Motor(Port.A, gears=gears)

    # Get expected and measured value.
    ratio = get_gear_ratio(gears)
    expected = real_angle / ratio
    measured = motor.angle()

    # Allow at most one degree of error, or up to 1/ratio when speeding up.
    tolerance = 1 if ratio >= 1 else 1 / ratio
    assert abs(motor.angle() - expected) <= tolerance, "{0} != {1} for ratio: {2}".format(
        expected, measured, gears
    )
```

laurensvalk, 2022-12-20:
This is done.

closed 2022-12-20 by laurensvalk

laurensvalk, 2023-01-10:
This now properly fixed by adding a close method in https://github.com/pybricks/support/issues/904