pybricks/support #565: duty with motor()control.limits

falk12, 2021-12-29:
Hi,

was there a change in the control.limits ?
It used to return 4 values, now only three?

Syntax and description are not consistent any more?

_control.limits(speed, acceleration, torque)¶
Configures the maximum speed, acceleration, duty, and torque._

laurensvalk, 2021-12-30:
> was there a change in the control.limits ?

Yes, see the [changed](https://github.com/pybricks/pybricks-micropython/blob/master/CHANGELOG.md#310c1---2021-11-19) section here.

> Syntax and description are not consistent any more?

The argument bullet list and descriptions thereof are correct. But duty should indeed be removed from the generic description.

Thanks for spotting this and letting us know!

[image]

For some applications, reducing the torque can be quite useful.

If you still want to change the "duty limit", you can set instead set the maximum voltage, like this:

```python
my_motor.settings(max_voltage=5000)  # Sets the limit to 5V
```

The new way of doing it ensures that the limit is independent of the battery type you use. Also, it makes `DCMotor` and `Motor` more consistent. Finally, it makes it more logical for motors that have a rated voltage limit, like the small motor. By just using the voltage, there is no confusion about what "100%" duty means.

closed 2021-12-30 by laurensvalk

falk12, 2021-12-30:
Thanks for support, so I will have to change and redistribute my programs.
All the best for 2022!