pybricks/support #566: [Question] How I can enable _thread on cityhub?

serkiri, 2021-12-29:
**Question**
How I can enable _thread on cityhub? 
I managed to compile cityhub firmware locally and programmed it to hub itself but I can't get how to include _thread package
I add 
#define MICROPY_PY_THREAD           (1)
to bricks\cityhub\mpconfigport.h
but 'make' gives me 
../../micropython/py/mpthread.h:38:10: fatal error: mpthreadport.h: No such file or directory
   38 | #include <mpthreadport.h>
 
Or it is not possible to use _thread with stm32 at all?

Thanks in advance and great job guys,

dlech, 2021-12-29:
Pybricks is not designed to use threads. See #356 for some ideas on how to make tasks that run in parallel without using threads.

serkiri, 2021-12-30:
What I am trying to reach is  - when I turn on vehicle it executes routine - starts to move and read sensors (cityhub) - it is better to do actions each 100ms. But at the same time if I turn on Remote (during routine) it stops and executes commands from Remote.

But the thing is I can't execute Remote(name, timeout) with timeout = 100ms, it works well only if timeout > 2000ms. But I can't afford routine to stop for 2s.

Is there any suggestion how to implement this? Now I thinking in the way to provide new API like NonBlockingRemote(name, timeout) with implementation in C which will provide getStatus() and then check it in the main routine if it CONNECTED or NOT_READY, or TIME_OUT.

Does it make any sense?

Regards,

laurensvalk, 2021-12-30:
> What I am trying to reach is - when I turn on vehicle it executes routine - starts to move and read sensors (cityhub) - it is better to do actions each 100ms. But at the same time if I turn on Remote (during routine) it stops and executes commands from Remote.

How about connecting to the remote at the start of your script?

You could then read the remote center button to switch between autonomous and remote control mode.

serkiri, 2021-12-30:
That will definitely work but there is no challenge in it at all )))

laurensvalk, 2021-12-30:
I'd probably go for the solution that "will definitely work" given the sheer number of other open challenges :smile: 

> But the thing is I can't execute Remote(name, timeout) with timeout = 100ms, it works well only if timeout > 2000ms. 

I can confirm that this does not work. I have opened an issue for it, referenced above.

serkiri, 2021-12-30:
You guys are awesome, thanks a lot for your work!

closed 2021-12-30 by serkiri