pybricks/support #567: [Bug] Powered up remote connection task can get stuck in loop

laurensvalk, 2021-12-30:
**Describe the bug**

When running the code below the hub gets stuck. It cannot be stopped with the hub button, nor with Pybricks Code stop button.

The hub light is still fading so `pbio` is still running.

**To reproduce**

Install firmware on City Hub with `Pybricks v3.1.0 (Pybricks Beta v1.2.0-beta.1)`.

Run this program from Pybricks Code.

After a few prints, turn on the remote. Then everything stops.

```python
from pybricks.pupdevices import Remote
from pybricks.tools import wait

for i in range(1000):
    print("Attempt", i)
    try:
        remote = Remote(timeout=100)
        print("Connected!")
    except OSError:
        pass
```

**Expected behavior**
This program should print `Attempt X` every 100 ms until the remote is turned on. Then it should print `"Connected!"`.

**Additional context**
Above example was inspired by the discussion _Originally posted by @serkiri in https://github.com/pybricks/support/issues/566#issuecomment-1003046198_

dlech, 2022-01-03:
Many Bluetooth tasks are currently not properly cancelable and don't have proper error detection. With this small of timeout, the Bluetooth chip probably doesn't even have time to start scanning. It probably would have been better if we would have left out the timeout parameter.

SpudGunMan, 2022-01-08:
I do find the timeout handy might just need a documentation details about this?

laurensvalk, 2022-11-14:
This also happens for the default 10 second timeout eventually.

closed 2023-04-01 by dlech