pybricks/support #574: [Bug] MS Inventor hub does not turn off light when program is stopped

lobodpav, 2022-01-01:
**Describe the bug**
MindStorms Inventor hub does not turn off lights of the UltrasonicSensor when program is stopped.

**To reproduce**
Steps to reproduce the behavior:
1. In your program, turn on UltrasonicSensor lights
2. Run the program on the Hub
3. Press the round button to stop the program

**Expected behavior**
The UltrasonicSensor lights should get turned off.

**Example program**
```python
from pybricks.pupdevices import UltrasonicSensor
from pybricks.parameters import Port

distance_sensor = UltrasonicSensor(Port.D)
distance_sensor.lights.on()

while True:
    continue
```

laurensvalk, 2022-01-02:
Thanks for opening this issue.

The color sensor lights stay on as well. Is there any reason why the ultrasonic sensor lights should be turned off?

lobodpav, 2022-01-02:
I would turn off everything. Also, the colour sensor is lit by default even after starting up the hub.

In my eyes, the current behaviour is
1. illogical: Stopping a program by the Hub's button shall turn all stuff off. It would also save power while nothing is running.
2. inconsistent: When a program is stopped by the hub's button, motors are turned off and the display is reset to the "cube image". However, the lights are left intact.
3. inconvenient: Once I turn the hub on, the colour sensor's light is on and stays on. And it shines a lot. The same applies to the Ultrasonic sensor's lights it turned to 100%. When working on a robot's program at night, it just blinds me so I need to put the robot face down which is inconvenient.

laurensvalk, 2022-01-03:
The color sensor behavior is currently consistent with the official MINDSTORMS app. Its light turn on by default and there is no state change when the program ends. We currently do the same for the ultrasonic sensor: no state change when the program ends.

The problem with turning the color sensor off completely is that it takes a while for the first readings to settle. If we turned it on just when the program begins, the first readings are not right. I haven't tested this for the ultrasonic sensor.

It's also possible to turn off the lights by sending a special message to the sensor, which is what happens when you use the `lights` attribute. If we wanted to do something like this, we'd need to have a way to run some code after the user code ends, and wait for that code (and uart messages) to finish sending.

Another way is to handle it in your programs, like this:

```python

def main_code():
    print("hi!")

try:
    main_code()
except SystemExit:
   # This runs when you stop the robot, so here you could turn off any lights

```

lobodpav, 2022-01-03:
Thanks for explaining!

Did not realise the Color light reading issue on start-up. It is just very annoying turning on the Hub and being dazzled by the light. Unfortunately, a program can't fix that when the hub turns itself off after some time and once I start it up again, boom 😄

The UltrasonicSensor lights are for aesthetics only I believe.

dlech, 2022-01-03:
Wouldn't it make more sense to use `finally`? Then `cleanup()` only needs to be called once.

lobodpav, 2022-01-03:
I think this deserves an example section in the documentation. https://github.com/pybricks/pybricks-api/pull/90

laurensvalk, 2022-01-07:
> I think this deserves an example section in the documentation.

Done via https://github.com/pybricks/pybricks-api/pull/90, so this can be closed.

closed 2022-01-07 by laurensvalk