pybricks/support #590: [Feature] run code on powerup /boot with no 'button press'

SpudGunMan, 2022-01-11:
a feature request for running code on power up possibly using the way the VScode extension did it such as parsing the first line of the .py file to look for pybricks instructions

`# PYBRICKS autorun=true`

or something like that to signify the .py program should start at boot. now this adds a support problem of stuck loop issue if your code is bad. so the way out would need to be a DFU/flash of the firmware which is acceptable IMO should have tested first

the use case I have is my kids cant seem to figure out how to press the button twice to use the trains so its a very dumb use case, but nonetheless here we are!

dlech, 2022-01-12:
The MicroPython way to do this would be to just have a file named `boot.py`.

dlech, 2022-01-12:
This was also requested in https://github.com/pybricks/support/issues/523#issuecomment-1002544884.

laurensvalk, 2022-01-13:
> the use case I have is my kids cant seem to figure out how to press the button twice to use the trains so its a very dumb use case, but nonetheless here we are!

If this is the use case, maybe we just need to improve the UI :)

Alternatively, how about making a program to manage this yourself? You can already do that with today's tools.

You can make a program that starts/stops motion when you press the button (without truly stopping the program), and control the lights accordingly. Effectively, this makes everything work with a single press.

If you make sure to turn the light off in the "off" state, the batteries should last a long time.

laurensvalk, 2022-01-13:
> This was also requested in [#523 (comment)](https://github.com/pybricks/support/issues/523#issuecomment-1002544884).

Reading between the lines, I think that was suggested as an idea to potentially let you load larger programs on Move Hub. Quote:

> Is there also a way to make a program autostart just to start the downloaded program right when powering up?
This way you could make "permanent final programs" which uses less memory as they are not intended for development anymore. If you need do develop a new program then you reinstall the full firmware and start the program with the play icon.

But removing the button click doesn't result in any substantial savings. We could technically build a dedicated firmware without Bluetooth for that use case, but that doesn't seem worth it as you wouldn't be able to test it with constant reflashing.

ZPhilo, 2022-01-13:
> 
> If you make sure to turn the light off in the "off" state, the batteries should last a long time.

Not so sure... Idle consumption of the hub is far from negligible - at least with LEGO firmware, you may have done better :D
https://github.com/pybricks/support/issues/590#issuecomment-1011945060

SpudGunMan, 2022-01-13:
> If this is the use case, maybe we just need to improve the UI :)
>
> Alternatively, how about making a program to manage this yourself? You can already do that with today's tools.
>
> You can make a [...]

Yea like I said silly use case, as you mentioned it’s what I do now, (status LeD)

 The kids can’t seem to press the button twice once for power  on once for program run (young children) it’s no big deal I can’t see needing this feature outside of small children problem, really and the kids will learn eventually so it’s a very low issue or can be closed if it adds complications.

MilanV, 2023-11-13:
Please, is this issue still alive? I have another use case for this: my hubs are not physically accessible at all when the model is at exhibition. Hubs are on power adapter and I need they simply start my program when the exhibition organizers turn the electricity on (like Arduino does). Is it possible somehow?

(Yes, I have to use AC/DC adapter instead of batteries. With a very thin and simple "connector" hold between batteries (as a placeholder here) and the hub contacts. This way I do not need any modification of the hub. No glue, no cutting... Just PyBricks inside :D )

laurensvalk, 2023-11-13:
That's not currently supported.

So far, the need to auto-start on AC power-on has mainly come from commercial parties, but none of these parties seem to want to make the step to sponsor this feature :)

SpudGunMan, 2023-11-13:
I opened this issue (I think) and I have been supporting the project since that day. This last post turned me off to continuing that sponsorship. Thank you to the prior comment for saving me money!

laurensvalk, 2023-11-13:
Sorry to hear that @SpudGunMan --- this wasn't directed at your use case but at several exhibition organizers (including some large commercial ventures) who have requested this feature. I should have clarified.

SpudGunMan, 2023-11-13:
Oh well, possibly one of them corporate will step up. I'm out. You can close this issue please.

laurensvalk, 2023-11-13:
Thanks for your input.

We try to prioritize those feature requests that help the most regular users. Here is a good example: https://github.com/pybricks/support/issues/989#issuecomment-1471754817

The auto-start feature was requested by only one regular user (commercial requests don't count), so it hasn't been on the top of our list. Sorry about that.

We appreciate all contributions from end users, but I'm afraid there isn't a way to directly link them to specific issues. If there was that impression, maybe we should try to clarify this somewhere.

> You can close this issue please.

Sure, no problem.

It can be re-opened if there is broader non-commercial interest in this feature.

closed 2023-11-13 by laurensvalk

MilanV, 2023-11-13:
OK. I could not imagine what I'd start - or finish, depends on the POV :)
Anyway, I understand priorities and I accept my use case is not mentioned by many people.
Unfortunately, it means a tough decision for me. I really love how PyBricks project works and makes  LEGO PowerUp hubs reliable and useful. And I know about the only alternative with start-on-poweron: raspberry Pi with the "Build HAT" - no original LEGO hub, no PyBricks :(

Isn't there any other way around. Like Arduino sending "some clever BT packet" or so?

laurensvalk, 2023-11-13:
Sure! You can use the Pybricksdev command utility (and library) to download and run new programs, and I think also to start an existing one.

You can do all of this without pressing any buttons on the hub.

MilanV, 2023-11-13:
Thank you for the tip! I check it.