pybricks/support #591: BETA: How to use Pybricks on SPIKE Prime and MINDSTORMS Robot Inventor

laurensvalk, 2022-01-14:
By popular demand, we are now releasing the first beta of Pybricks for SPIKE Prime and MINDSTORMS Robot Inventor!

Try it out now! :tada: 

[[image]](https://install.pybricks.com/)

### How can I try Pybricks for SPIKE Prime and MINDSTORMS Robot Inventor?
1. Follow the steps [on this page](https://install.pybricks.com/) to install Pybricks.
    - This is a temporary page to simplify the installation process.
    - Eventually this will just be done via our [main app](https://beta.pybricks.com/). Please bear with us :-)
2. [Start coding](https://beta.pybricks.com/). It's the same as for the other hubs.
3. To restore the original firmware, see the [instructions](https://install.pybricks.com/) on the installation page.

### What's working?

- Write and run code via the [Pybricks](https://beta.pybricks.com/) editor.
- The program is saved on the hub when you run it. You can start it again with the center button.
- Faster program execution, and lots of RAM available for your scripts.
- Battery charging and low battery indicators are now working.
- You can access most hub features via the [`PrimeHub` and `InventorHub`](https://docs.pybricks.com/en/latest/hubs/primehub.html) classes.
- You can use [all motors](https://docs.pybricks.com/en/latest/pupdevices/index.html), including SPIKE, MINDSTORMS, Technic and Powered Up.
- You can use [all sensors](https://docs.pybricks.com/en/latest/pupdevices/index.html), including SPIKE, MINDSTORMS, Technic and Powered Up.
- You can use the [Powered Up Remote Control](https://docs.pybricks.com/en/latest/pupdevices/remote.html).
- And [everything else listed in the docs](https://docs.pybricks.com/en/latest/index.html): the `pybricks` API is the same for all hubs.
- Advanced users can also run programs from the command line or their favorite editor using [pybricksdev](https://github.com/pybricks/pybricksdev).

### What's still in progress?
- Automatic firmware updates within our [main app](https://beta.pybricks.com/). (V3.2)
- Editing and saving multiple scripts. (V3.2)
- Creating scripts with reusable modules. (V3.3 or later)
- Better sound support and better IMU support. (V3.3 or later)
- Hub-to-hub communication. (V3.3 or later)

### What's the difference between the Prime Hub and the Inventor Hub?
- The color of the plastic casing.

### How can I help?

Give it a try, and let us know what you think!

- If you think you've found a bug, please [open a new issue](https://github.com/pybricks/support/issues).
- If you have any questions, please [start a new discussion](https://github.com/pybricks/support/discussions).
- If you just want to say "Yes, it works!!" please feel free to leave a comment here! :smile: 
- If you want to sponsor Pybricks, [please have a look here](https://github.com/sponsors/pybricks).

### What if I tried the alpha version previously?

No additional action is required. Just use the official SPIKE or MINDSTORMS app to install the latest official LEGO firmware, just like everybody else. Then you can run the Pybricks installer as indicated above. You may need to refresh the page to see the latest version.

Unlike the old alpha version, this beta no longer uses dual-boot to run Pybricks. Since restoring the original firmware takes just 30 seconds anyway, it's much simpler to just have one firmware. This way, all hubs work the same.

And of course, thanks a lot for trying this alpha and beta version! We really appreciate it.

DanNetor, 2022-01-14:
Hy @ laurensvalk,

nice job, thanks a lot. following questions / annotations from my side:

Q1:
This isn't a dual-boot installation, correct? Are you still planning to create a dual-boot installation?

A1:
As soon as a process is running on the hub (e.g. because the official loge MINDSTORMS / SPIKE app is still open (connection attempts)) the following error appears when clicking on "connect":
--> Failed to connect. Please wait 60 seconds and try again. Make sure that no other apps are using the hub."
The entire browser crashes (Edge, Windows 11). There are no entries in the Windows "event log"

laurensvalk, 2022-01-14:
That's right, see the post above regarding single/dual boot.

Thanks for reporing the connection issue. The error message is correct (i.e. you have to close the mindstorms app), but I'll double-check the browser freeze.

laurensvalk, 2022-01-14:
Thanks for that as well. Yes, it will work without a cable too. The cable just ensures that it restarts automatically. I'll update the script to pause until you plug it in.

laurensvalk, 2022-01-14:
I'll be moving your comment to a separate issue so we can keep this thread easy to follow for new users.

BertLindeman, 2022-01-14:
Did the install on win11 pro. Fine.

Ran a program using the IMU with tilt, acceleration and angular_velocity.
Did look OK.

Thanks!

jriise, 2022-01-14:
Does not work on FW 1.4.01. Do you know where I can find 1.3.00? I used to use the firmware suggested in dfu.pybricks.com, but that has been updated to 1.4.01 as well..

johnscary-ev3, 2022-01-14:
This is working great for me!
Now that the battery charging is included, the dual booting is not needed as much.
Although it was sort of cool.
The saved program run option is also a nice addition,
Thanks for getting this running.

SpudGunMan, 2022-01-15:
no bueno here, install-dot crashed ungoogled chromium once but I thought this could work with newest firmware? do I still need to downgrade my hubs to put this on?

```
Getting firmware info.
Your hub runs LEGO Firmware version: b'v1.4.01.0000-594ce3d'
Your LEGO firmware version is not (yet) supported.
Please install one of the following versions and try again.
```

dlech, 2022-01-15:
You need to clear your browser cache to get the latest page (opening in private window also works).

SpudGunMan, 2022-01-15:
oh yea that helped. for sure the issue, worked perfect

Nurech, 2022-01-16:
Following instructions to restore Inventor HUB fails. Althought the program itself ran very nice, ty :) 

```python
https://beta.pybricks.com/

from pybricks.experimental import restore_firmware
restore_firmware()

Traceback (most recent call last):
  File "main.py", line 2, in <module>
OSError: [Errno 1] EPERM: 

The requested operation is not valid in the current state:
--> Check the documentation for required conditions.
--> Check the line in your script that matches
    the line number given in the 'Traceback' above.

```

hamptonmoore, 2022-02-17:
Any update on hub to hub communication?

kmaar, 2022-02-24:
Thanks for your awesome work! I had Pybricks for my new Spike Prime working within 10mins. You rock! 🔥 🔥 🔥

hamptonmoore, 2022-02-25:
Got Spike with the Remote Control 88010 working wonderfully! Our students are very happy to finally have remote controllable Spike bots! Thank you

dlech, 2022-07-28:
It is now possible to flash firmware to SPIKE and MINDSTORMS (RI-only) hubs in Pybricks Code v3.2.0b3 (Pybricks Beta v2.0.0-beta.3).

closed 2022-07-28 by dlech

BertLindeman, 2022-07-28:
Great work, David.

I only had to update my (Win11) usb driver for SpikePrime hub.
Firmware updates for various hubs now neatly in one place.
Especially Spike and Mindstorms hubs blazing fast.

Thank you Pybricks team.