pybricks/support #592: SPIKE USB installer crashes the browser if another app is using the hub

laurensvalk, 2022-01-14:
As soon as a process is running on the hub (e.g. because the official loge MINDSTORMS / SPIKE app is still open (connection attempts)) the following error appears when clicking on "connect":
--> Failed to connect. Please wait 60 seconds and try again. Make sure that no other apps are using the hub."
The entire browser crashes (Edge, Windows 11). There are no entries in the Windows "event log"

_Originally posted by @DanNetor in https://github.com/pybricks/support/issues/591#issuecomment-1012967894_

laurensvalk, 2022-01-14:
> Failed to connect. Please wait 60 seconds and try again. Make sure that no other apps are using the hub

This is the expected behavior. All other apps that use the hub must be closed.

> The browser(s) (also affects chrome) do not freeze, they simply crash.

I can't seem to reproduce this unfortunately.

dlech, 2022-01-14:
If the entire browser crashes, then the issue needs to be reported to Microsoft. There is nothing we can do about it.

laurensvalk, 2022-09-01:
We are now using DFU for installation, so we can close this.

closed 2022-09-01 by laurensvalk