pybricks/support #597: Robot Inventor installer does not work for FW 1.4.01[Bug]

jriise, 2022-01-14:
The installer does not work for FW 1.4.01

Connection process and downloading the new firmware is OK, but during installation it says that the firmware is too new.

When I've tried pybricks on my Robot Inventor earlier, I've used to use the firmware file in dfu.pybricks.com to flash the brick prior to installation, but this has also been changed to 1.4.01.

laurensvalk, 2022-01-14:
I think you might need to refresh the page. Try CTRL+F5 if just refreshing doesn't work.

Your version is supported:
https://github.com/pybricks/pybricks-micropython/blob/master/bricks/stm32/install_pybricks.py#L64

jriise, 2022-01-14:
Got it working using in private browser window. I had this page stored as a favorite, maybe that caused the issue.. Anyway, great work!. Really appriciate the time and effort put in to this project!

closed 2022-01-14 by dlech