pybricks/support #598: [Bug] Clean install error for Mindstorm RI Hub

poldolot, 2022-01-14:
During installation on MacOS, Chrome browser I've got this error:
```
Something went wrong. Please try again.
from _firmware.install_pybricks import install; install()
Starting installation script.
Checking installation files.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "_firmware/install_pybricks.py", line 151, in install
  File "_firmware/install_pybricks.py", line 134, in get_file_hash
OSError: [Errno 2] ENOENT
>>> 
```

laurensvalk, 2022-01-15:
This looks like #597. Can you hard refresh the page?

poldolot, 2022-01-15:
I'm php programmer, so I do hard refresh every time, but to be sure I've also tried on second MacBook and the installation went through with no error, that's weird ;>

closed 2022-01-15 by poldolot

laurensvalk, 2022-01-15:
If you want to take a screenshot of the page, it should be easy to find out.

Or maybe it had refreshed the page, but not the resources.

laurensvalk, 2022-01-15:
This will all be taken care of when it's integrated in our main app, which checks for the right versions. :)

Sorry for the trouble.