pybricks/support #599: [Bug] primehub incompatible .mpy file on center press

SpudGunMan, 2022-01-15:
when you have no code running and press the center button on new prime hubs the console displays 

`ValueError: incompatible .mpy file`

**Expected behavior**
either program run or no output, functionality like the OEM program slot and run is I think in the works and this might be related to that

found an old issue to link it to https://github.com/pybricks/support/issues/290

also this one is related https://github.com/pybricks/support/issues/590

dlech, 2022-01-15:
Hubs with external flash currently don't have a preloaded `main.py` like other hubs. After you download and run a program and stop it, then pressing the button should run that program again.

laurensvalk, 2022-01-15:
This currently happens if there's no program yet.

johnscary-ev3, 2022-08-23:
Hi guys,

I just upgraded to firmware pybricks-primehub-v3.2.0b3.zip  from v3.2.0b1
That went ok but now I get this error with same program that used to work.
The program builds the .mpy file ok and downloads it but then I get the "incompatible .mpy file" error.
My system is running Python 3.10.6 which I believe makes the .mpy file for us.
I noticed that one of the recent updates upgraded the micro-python version.
Could that be a problem?  Any ideas?

```
PS C:\Users\johns\pybricksdev-demo>  c:/Users/johns/pybricksdev-demo/.venv/Scripts/python.exe c:/Users/johns/pybricksdev-demo/RemoteControl_Blast_New_June_2022.py
main: Start
forwarder: Start
100%|███████████████████████████████████████████████████████████████| 40.5k/40.5k [01:14<00:00, 542B/s]
forwarder: Hub Running
ValueError: incompatible .mpy file
main: Stop
```

dlech, 2022-08-23:
Make sure you are running the latest `pybricksdev` (1.0.0a29 ) in your virtual environment.

johnscary-ev3, 2022-08-23:
Thanks,
I updated from pybricksdev 1.0.0a27 to 1.0.0a29 and that fixed the .mpy file problem.
Interestingly I got another error due to this line "from math import pi" in my robot program which was ok before.
I fixed that problem, and my program is running ok now.
Appreciate the quick reply on this.

laurensvalk, 2022-09-01:
This issue was fixed in https://github.com/pybricks/pybricks-micropython/pull/110 but it didn't seem to have closed automatically. Closing it now.

closed 2022-09-01 by laurensvalk