pybricks/support #600: Use random Bluetooth address

dlech, 2022-01-16:
**Is your feature request related to a problem? Please describe.**
We are currently using the same Bluetooth address as LEGO for all hubs but we have a different GATT database. This causes problems when the OS that connects to the hub caches GATT info. This results in errors about missing GATT services/characteristics and such. BlueZ is the worst about this and usually requires manually removing the device and manually deleting a cache file.

I was also just chatting with someone who had to reboot their iPhone in order to get the official LEGO apps to work again because connecting to the hub after restoring LEGO firmware was not working. Not sure if this is related, but sounds similar.

**Describe the solution you'd like**
If we use a random Bluetooth address on every connection, then the OS will see the hub as a new device every time and we can avoid issues with wrong cached values.

**Describe alternatives you've considered**
Keep things the way they are.

SpudGunMan, 2022-01-18:
do you need a new one each time? why wouldn't you just make a new one for pybricks just offset the original? makes for more of an identification standard for multiple devices

dlech, 2022-01-19:
> do you need a new one each time?

This would avoid many issues with caching on Linux (and possibly other OSes, but that has yet to be proven).

> why wouldn't you just make a new one for pybricks just offset the original?

Apple doesn't allow access to the Bluetooth address for security/privacy reasons. So having a fixed address is not useful on all platforms. Instead, we prefer using the device name as the unique identifier. It works on all platforms and is more memorable and meaningful than a Bluetooth address.

Novakasa, 2022-01-21:
This would impact a usecase of mine in which I automatically keep track of different hubs to store information about them on PC. I understand that I can use names for this, but then this would require the users of my project to set unique names for each of the hubs. It would be nice if we had a way to uniquely identify hubs regardless of the names. I think even just providing unique default names for the hubs could be helpful.

dlech, 2022-01-21:
Does the unique identifier need to be in the advertising data (before connecting) or is reading it after connecting good enough? https://github.com/pybricks/pybricks-micropython/issues/50 has some more ideas on this.

Novakasa, 2022-01-21:
It would be ideal if it were in the advertising data, since otherwise upon connecting a specific one I would need to trial-connect each hub.

Novakasa, 2022-05-11:
After rethinking the requirements for my projects, I think the name as implemented now should suffice for me after all.

dlech, 2022-10-21:
Implemented via nonresolvable private addresses in https://github.com/pybricks/pybricks-micropython/commit/1815b8e6b65ca46ae20a9b41dc9b98a4b22735d4.

laurensvalk, 2022-10-21:
When disconnecting and connecting, the previously used hub lingers for a while.

[image]

Attempting to connect can fail in one of two ways. Either you get an infinite loop in Pybricks Code:

[Screencast from 21-10-22 09:59:09.webm](https://user-images.githubusercontent.com/12326241/197144430-70cd97cb-e085-49aa-8cbb-3237038efbd0.webm)

Or you get this error:

[image]

laurensvalk, 2022-10-21:
It sticks around for just a few seconds, so it isn't terrible.

I know we can't change how the Web BLE UI works, but perhaps we could make a more informative error message if the newly selected hub has an identical address, and tell them to try again.

By the time they try again, the old hub will be gone anyway.

laurensvalk, 2022-10-21:
Side note: it fixes switching between pybricksdev  and pybricks code as expected. Nice!

dlech, 2022-10-21:
> It sticks around for just a few seconds (on Ubuntu 22.04 + Chrome), so it isn't terrible.

This behavior is only seen on Linux and would need to be fixed in BlueZ and/or Chromium, so I think we can call it good and close this issue.

laurensvalk, 2022-10-21:
Getting out of the infinite loop in Pybricks Code might be worth doing though?

dlech, 2022-10-21:
Hmm... I missed that one. Will look into it.

closed 2022-10-26 by dlech

laurensvalk, 2022-11-26:
On Move Hub and Technic Hub, I always see the same Bluetooth address in the connection dialog. Even after rebooting the hub. Is this expected?

To reproduce, have both Technic Hub and Move Hub and a Prime Hub for comparison, all named "Pybricks Hub" so Chrome will show the addresses.

Only the address for Prime Hub changes after rebooting the hub and re-opening the dialog.

dlech, 2022-11-26:
I get a new address after reboot on both hubs using Bleak, so maybe a Chrome bug?

```
(bleak-py3.10) david@freyr:~/work/pybricks/bleak$ python examples/discover.py 
scanning for 5 seconds, please wait...

18:10:6A:CB:8D:8E: technic_hub
---------------------------
AdvertisementData(local_name='technic_hub', service_data={'00002a50-0000-1000-8000-00805f9b34fb': b'\x01\x97\x03\x80\x00\x00\x00'}, service_uuids=['c5f50001-8280-46da-89f4-6d8051e4aeef'], tx_power=0, rssi=-57)

0C:95:4D:80:FA:6D: move_hub
---------------------------
AdvertisementData(local_name='move_hub', service_data={'00002a50-0000-1000-8000-00805f9b34fb': b'\x01\x97\x03@\x00\x00\x00'}, service_uuids=['c5f50001-8280-46da-89f4-6d8051e4aeef'], tx_power=1, rssi=-47)
(bleak-py3.10) david@freyr:~/work/pybricks/bleak$ python examples/discover.py 
scanning for 5 seconds, please wait...

19:7F:18:36:15:49: move_hub
---------------------------
AdvertisementData(local_name='move_hub', service_data={'00002a50-0000-1000-8000-00805f9b34fb': b'\x01\x97\x03@\x00\x00\x00'}, service_uuids=['c5f50001-8280-46da-89f4-6d8051e4aeef'], tx_power=1, rssi=-56)

30:17:F7:3C:06:20: technic_hub
---------------------------
AdvertisementData(local_name='technic_hub', service_data={'00002a50-0000-1000-8000-00805f9b34fb': b'\x01\x97\x03\x80\x00\x00\x00'}, service_uuids=['c5f50001-8280-46da-89f4-6d8051e4aeef'], tx_power=0, rssi=-65)
```

laurensvalk, 2022-11-26:
I think I'm only able to reproduce it when my PC (workstation) and Laptop are both scanning in Pybricks Code at the same time.

I think I'm on to something --- it seems I can finally reproduce some problems I saw at WRO last week (presumably with a lot of laptops around):

- Hub not finding remote
- Technic hub stuck in the power off blink loop