pybricks/support #610: [Bug] 'Charger' object has no attribute 'connected'

DanNetor, 2022-01-19:
**Describe the bug**
No access to the "connected" attribute in the "charger" object

**To reproduce**
Following code leads to the error:
`
from pybricks.hubs import PrimeHub

hub = PrimeHub()
print('     ---> connected: ', hub.charger.connected())

raise SystemExit
`

**Expected behavior**
The method returns a boolean value indicating whether the charger is connected (USB) or not. According to documentation:
https://docs.pybricks.com/en/latest/hubs/primehub.html

dlech, 2022-01-19:
This was just implemented a day or two ago, so unless you compiled your own firmware, it probably isn't there yet.

laurensvalk, 2022-01-20:
Thanks for your note. I've opened a new issue in #612 to clarify what each version of the documentation is and how to change between them.

closed 2022-01-20 by laurensvalk

DanNetor, 2022-01-20:
@dlech:
I used the original firmware from [the pybricks installer](https://install.pybricks.com/).

So i think that is the latest one?

> This was just implemented a day or two ago, so unless you compiled your own firmware or flashed the latest build from GitHub actions, it probably isn't there yet.

laurensvalk, 2022-01-20:
We release new versions via the Pybricks installer only if they have been tested, just to make sure that the core are working.

If you want the bleeding-edge version, you'd have to install the firmware yourself or build it from source, or just wait a week or two :)

DanNetor, 2022-01-20:
ok, now i understand the topic. Just a "timing" problem between documentation and firmware release-process.
thanks all!