pybricks/support #611: [Feature] Absolute angle for motors

jriise, 2022-01-19:
**Is your feature request related to a problem? Please describe.**
When building walking robots like Gelo, it would be nice to use absolute angles for the walking motion. For walkers, the legs need to be syncronized to each other for a smooth walk (I've used StopWatch as a value generator for track_target). The problem arises when you turn the robot by walking one or two legs in the other direction and these start accumulating a very high angular values in the other direction. For now, the solution for me is to stop the motion and reset to absolute values before continuing with a new direction of motion. This is fine for pre-programmed walk, but kills a bit of the joy, when using remote controll  

**Describe the solution you'd like**
Make motor commands that support the use of absolute angles 

**Describe alternatives you've considered**
If statements to reset motor angle when it exceeds + or - 180 degrees:
 -Causes glitchy behaviour in the motor motion

Exiting a walking state (Left turn) and resetting the motor angles prior to comencing a new walking state (forwards):
 -Works, but is a bit akward when using a remote to control the robot

Tried using iodevices class in pybricks as this looked a bit like using the hub class in mindstorms, but was not able to instatiate the Motor class.

**Additional context**
I've tried using the hub API using import hub (not MSHub) in the mindstorms inventor software and can make it work there (without the 88010 remote though). I still prefer pybricks due to the awsome level of documentation and easy programming enviroment. When using the hub API i can use the get() command to return a tuple containing, among other things, the absolute motor angle (speed_pct, rel_pos, abs_pos, pwm = hub.port.A.motor.get() ).

laurensvalk, 2022-01-19:
> Make motor commands that support the use of absolute angles

Could you describe what you would like to see? What would the command look like, and how would it make remote control better?

The motor angles returned by Pybricks already correspond to the visual marker on the shaft. If you prefer to read the value within a 360-degree circle, you could use the modulo operator (angle % 360).

jriise, 2022-01-19:
Sorry, I'm a noob, didn't know the modulo operator. I'll check that out.. Basically I'm looking for a readout between -180 and 179 (or degrees of a 360 degree circle, as you point out). The problem is not in the remote it self, but the interaction with the robot when angles keep spinning into plus and minus infinity when making a turn. 

In the hub API the command to get absolute angle looks like this: hub.port.A.motor.get([2]). The complete tuple contains speed, relative position, absolute position and pwm.  When using these low level commands, everything is super responsive and fast. But, since this would only work on spike and RI, it might be too much bother to implement it..

https://lego.github.io/MINDSTORMS-Robot-Inventor-hub-API/class_motor.html

Anyways, thanks for the tip about the modulo operator

laurensvalk, 2022-01-20:
To get a value between 0 to 360 you can do:

```python
motor.angle() % 360
```

To get it between -180 and 179, you'd have to shift it by -180, so:

```python
(motor.angle() + 180) % 360 - 180
```

I think we can add both of these as examples to the documentation.

If other people really want this too, we could add it to the existing API like this:

```python
motor.angle()   # Default behavior. Angle corresponds to marker, and keeps counting.

motor.angle(wrap=True)  # Wraps around back to -180 if the angle exceeds 180.
```

> I still prefer pybricks due to the awsome level of documentation and easy programming enviroment.

Glad you like it! :smile:

closed 2022-01-20 by laurensvalk

laurensvalk, 2022-01-20:
I've gone ahead and added an example here:
https://docs.pybricks.com/en/latest/pupdevices/motor.html#getting-the-absolute-angle

If anyone would rather see that functionality built-in, feel free to re-open the issue and argue your case :)

jriise, 2022-01-20:
Thank you so much for your help! This is an awsome project!