pybricks/support #612: [Doc] Clarify which versions to use

laurensvalk, 2022-01-20:
**Describe the issue**
We have `stable` docs (corresponding to the latest full release) and `latest` docs (corresponding to the latest `master` build).

This is a bit unclear, and leads to questions like https://github.com/pybricks/support/issues/610.

**Suggested improvement**
In the IDE, we have the following note (actually not right now, see below), but we should probably have something like this in the [online version](https://docs.pybricks.com/en/latest/) too.

[image]

To complicate matters, the latest beta is currently also the latest stable version, so nothing is shown at all. This should fix itself as soon as we make a new beta.

dlech, 2022-01-20:
It would also be very useful to add a note to individual APIs as to which version they were added in. Then we can know right away if something is new or not without having to dig into the source code. This also applies to changes in behavior, like some of the recent semi-breaking changes to motors.

The official Python docs do it like this:
[image]

laurensvalk, 2022-01-30:
When we look at this, here's a related one.

> Also, latest code editor (non-beta) is marked as v3.1 and when I select v3.1 in dedicated [documentation page](https://docs.pybricks.com/en/v3.1.0/hubs/technichub.html), IMU section is there.

_Originally posted by @estoykov in https://github.com/pybricks/support/issues/624#issuecomment-1025206099_