pybricks/support #613: [Question] Multitasking

MartinGarciaMatienzo, 2022-01-20:
**Question**
Is there a way of using "wait=False" an amount infinite of times or another funtion for parallel tasks?

**Context**
I was coding a robot using a technic hub. The robot consists in 4 motors, which 3 of them have to move at the same time in different directions and speeds continuously. So, if I try to use wait=False for the different stages of moving, it won't work. I was wondering if there was a way or trick to use this many times or another function. Maybe it´s something basic, I know

Here is part of the code:

from pybricks.hubs import TechnicHub
from pybricks.pupdevices import Motor
from pybricks.parameters import Port, Color, Button, Stop
from pybricks.tools import wait
from pybricksdev import 
#DECLARACION DE COMPONENTES
cerebro1=TechnicHub()
cerebro1.system.set_stop_button(Button.CENTER)
rojo=Motor(Port.A)
azul=Motor(Port.B)
amarillo=Motor(Port.C)
verde=Motor(Port.D)
#DEMO
rojo.run_time(500, 350, wait=False)
azul.run_time(500, 350, wait=False)
amarillo.run_time(500, 350, wait=False)
wait(1000)
verde.run_time(450, 2000)
if verde.control.done():
    rojo.run_time(-500, 350, wait=False)
azul.run_time(-500, 350, wait=False)
amarillo.run_time(-500, 350, wait=False)

dlech, 2022-01-20:
Yes. Have a look at https://github.com/pybricks/support/discussions/356.

laurensvalk, 2022-01-20:
Depending on what you need, the following solution may be a bit simpler:

https://docs.pybricks.com/en/stable/pupdevices/motor.html#waiting-for-two-parallel-actions-to-complete

MartinGarciaMatienzo, 2022-01-20:
I found this example avaible for EV3 does it work for the technic hub Motors? 

        from pybricks.experimental import run_parallel
        from pybricks.tools import wait

        def task1():
            wait(1000)
            return 'OK1'

        def task2():
            wait(500)
            return 'OK2'

        result = run_parallel(task1, task2)
        print('task1:', repr(result[task1]))
        print('task2:', repr(result[task2]))
        # prints:
        # task1: 'OK1'
        # task2: 'OK2'

dlech, 2022-01-20:
No, the EV3 has experimental support for threads, but the newer hubs don't support threads.

MartinGarciaMatienzo, 2022-01-20:
So is there a way to re-use wait=False in some kind of closed stages? The parallel task for running multiple Motors didn´t work well :c

dlech, 2022-01-20:
What is the expected behavior and what is the actual behavior of the code that you shared?

MartinGarciaMatienzo, 2022-01-20:
Well, it controls a delta robot so it uses 3 motors to incline the arms that hold the tools. The Demo raises and lowers the tool, therefore all the motors have to run at the same time in one direction, then, all of them 3 run in the opposite direction to go down. That is the Demo, seeing that it didn't work, I didn't continue with the rest. I first planned to control multiple motors in divided sections to get different angles and distances of the tool

SpudGunMan, 2022-01-21:
the co-routine might be applicable? https://gist.github.com/dlech/fa48f9b2a3a661c79c2c5880684b63ae

dlech, 2022-01-21:
yes, that is what #356 is

laurensvalk, 2022-01-21:
You can extend this for any number of motors.

```python
from pybricks.pupdevices import Motor
from pybricks.parameters import Port
from pybricks.tools import wait

# Initialize motors on port A and B.
track_motor = Motor(Port.A)
gripper_motor = Motor(Port.B)

# Make both motors perform an action with wait=False
track_motor.run_angle(500, 360, wait=False)
gripper_motor.run_angle(200, 720, wait=False)

# While one or both of the motors are not done yet,
# do something else. In this example, just wait.
while not track_motor.control.done() or not gripper_motor.control.done():
    wait(10)

print("Both motors are done!")
```

But in your example, you use `run_time`, so you could also just do this:

```python
DURATION = 350
rojo.run_time(500, DURATION, wait=False)
azul.run_time(500, DURATION, wait=False)
amarillo.run_time(500, DURATION, wait=False)
wait(DURATION)
```

MartinGarciaMatienzo, 2022-01-21:
Yes, I did that, but i couldn't run these 3 motors at the same time in reverse after this instruction to raise the tool in the center of the robot

MartinGarciaMatienzo, 2022-01-21:
I could, thanks, i used def to separate every action as dlech said. Thanks for your support :)

dlech, 2023-03-21:
Future visitors will be interested in the latest developments #917.

closed 2023-03-21 by dlech