pybricks/support #616: [Question] How to call a function on a running program?

Nurech, 2022-01-24:
Either directly or changing some variable to trigger such action. Essentially sending upstream data to a running program. Is this possible? If so, which documentation covers this?

dlech, 2022-01-25:
Please see https://github.com/pybricks/support/issues/262.

Nurech, 2022-01-25:
So it's in the works. Good luck! Closing this for now.

closed 2022-01-25 by Nurech

Novakasa, 2022-01-25:
In the meantime, maybe something like this is what would work for you: https://github.com/pybricks/support/issues/470#issuecomment-917070340

Nurech, 2022-01-26:
> In the meantime, maybe something like this is what would work for you (communicating with the hub via the `pybricksdev` python library): [#470 (comment)](https://github.com/pybricks/support/issues/470#issuecomment-917070340)

Thank you! At first glance that looks very promising.

laurensvalk, 2022-10-25:
There is now a tutorial here: https://pybricks.com/projects/tutorials/wireless/hub-to-device/pc-communication/

[image]

So we can close this issue.