pybricks/support #617: [Bug] Make deploy fails on zipfile dependency

laurensvalk, 2022-01-24:
When a SPIKE hub runs the stock firmware and you run

```
make -C bricks/primehub deploy -j36
```

then this should install Pybricks.

However, since https://github.com/pybricks/pybricks-micropython/commit/b168c00b02fe257e84d7f2456d8b3330cab12008 this is broken with

```
Creating firmware...
Connecting to /dev/ttyACM0
Connected!
Hub is ready.
Traceback (most recent call last):
  File "/home/laurens/git-pybricks/pybricks-micropython/.venv/bin/pybricksdev", line 8, in <module>
    sys.exit(main())
  File "/home/laurens/git-pybricks/pybricks-micropython/.venv/lib/python3.8/site-packages/pybricksdev/cli/__init__.py", line 374, in main
    asyncio.run(subparsers.choices[args.tool].tool.run(args))
  File "/home/laurens/.pyenv/versions/3.8.2/lib/python3.8/asyncio/runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "/home/laurens/.pyenv/versions/3.8.2/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "/home/laurens/git-pybricks/pybricks-micropython/.venv/lib/python3.8/site-packages/pybricksdev/cli/flash.py", line 309, in flash_firmware
    archive = zipfile.ZipFile(firmware)
  File "/home/laurens/.pyenv/versions/3.8.2/lib/python3.8/zipfile.py", line 1269, in __init__
    self._RealGetContents()
  File "/home/laurens/.pyenv/versions/3.8.2/lib/python3.8/zipfile.py", line 1332, in _RealGetContents
    endrec = _EndRecData(fp)
  File "/home/laurens/.pyenv/versions/3.8.2/lib/python3.8/zipfile.py", line 264, in _EndRecData
    fpin.seek(0, 2)
AttributeError: 'bytearray' object has no attribute 'seek'
../stm32/stm32.mk:687: recipe for target 'deploy' failed
make: *** [deploy] Error 1
make: Leaving directory '/home/laurens/git-pybricks/pybricks-micropython/bricks/primehub'
```

It looks like one of our dependencies has changed or we are using them incorrectly.

dlech, 2022-01-25:
`zipfile` is part of the standard library and is called directly from our code, so I don't think this could be a dependency issue other than on the Python version. I see that the stack trace is calling some code from the venv and some code from pyenv, so maybe you just need to wipe out the venv and start over?

laurensvalk, 2022-01-25:
> wipe out the venv and start over?

That's what this is. With such clean installs, the problem starts specifically at the aforementioned commit.

I'm on Ubuntu 18.04, so I have to provide Python 3.8 via pyenv to get poetry to even run.

I think I'll just use DFU until Ubuntu 22.04 comes out.

closed 2022-01-25 by laurensvalk

dlech, 2022-01-25:
Not sure why I'm not seeing this crash but there is an obvious mistake in the refactoring done in pybricksdev.

laurensvalk, 2022-01-25:
Thanks for the quick fix!