pybricks/support #618: [Bug] from pybricksdev import connections fails when debugging in vscode

Novakasa, 2022-01-25:
**Describe the bug**
When trying to debug this file in vscode:
```python
from pybricksdev import connections
```
I get the exception:
```
> Exception has occurred: RuntimeError
Cannot change thread mode after it is set.
  File "E:\repos\brickrail\ble-server\test_hub_script.py", line 1, in <module>
    from pybricksdev import connections
```
When running the command "Run Python File in terminal", the error does not occur. I am guessing vscode does some thread shenanigans that some dependency of pybricksdev is not happy with. If I'm not crazy, this has worked before last summer or so)

**To reproduce**
- Open vscode
- Enter the following script:
```python
from pybricksdev import connections
```

**Expected behavior**
No exception. This has worked in the past with previous versions.

**Environment**
```
PS E:\repos\brickrail> pip list
Package                Version
---------------------- -------------------
aioserial              1.3.0
appdirs                1.4.4
argcomplete            1.12.3
argon2-cffi            20.1.0
async-generator        1.10
asyncssh               2.7.1
attrs                  21.2.0
backcall               0.2.0
bleach                 3.3.0
bleak                  0.14.1
bleak-winrt            1.1.1
brotlipy               0.7.0
certifi                2021.5.30
cffi                   1.14.4
chardet                3.0.4
colorama               0.4.4
conda                  4.10.3
conda-package-handling 1.7.2
cryptography           3.3.1
cycler                 0.10.0
decorator              5.0.9
defusedxml             0.7.1
entrypoints            0.3
idna                   2.10
importlib-metadata     3.10.0
ipykernel              5.3.4
ipython                7.22.0
ipython-genutils       0.2.0
ipywidgets             7.6.3
jedi                   0.17.2
Jinja2                 3.0.1
jsonschema             3.2.0
jupyter                1.0.0
jupyter-client         6.1.12
jupyter-console        6.4.0
jupyter-core           4.7.1
jupyterlab-pygments    0.1.2
jupyterlab-widgets     1.0.0
kiwisolver             1.3.1
MarkupSafe             2.0.1
matplotlib             3.3.4
menuinst               1.4.16
mistune                0.8.4
mkl-fft                1.3.0
mkl-random             1.2.1
mkl-service            2.3.0
mpy-cross              1.14
nbclient               0.5.3
nbconvert              6.1.0
nbformat               5.1.3
nest-asyncio           1.5.1
notebook               6.4.0
numpy                  1.20.2
olefile                0.46
packaging              21.0
pandocfilters          1.4.3
parso                  0.7.0
pickleshare            0.7.5
Pillow                 8.3.1
pip                    20.3.1
prometheus-client      0.11.0
prompt-toolkit         3.0.20
pybricks               2.0.0.post1
pybricksdev            1.0.0a24
pycosat                0.6.3
pycparser              2.20
Pygments               2.9.0
pyOpenSSL              20.0.0
pyparsing              2.4.7
pyrsistent             0.18.0
pyserial               3.5
PySocks                1.7.1
python-dateutil        2.8.1
pyusb                  1.2.1
pywin32                228
pywinpty               0.5.7
pyzmq                  20.0.0
qtconsole              5.1.0
QtPy                   1.9.0
requests               2.25.0
ruamel-yaml-conda      0.15.80
semver                 2.13.0
Send2Trash             1.5.0
setuptools             51.0.0.post20201207
sip                    4.19.13
six                    1.15.0
terminado              0.9.4
testpath               0.5.0
tornado                6.1
tqdm                   4.62.3
traitlets              5.0.5
urllib3                1.25.11
validators             0.18.2
wcwidth                0.2.5
webencodings           0.5.1
wheel                  0.36.1
widgetsnbextension     3.5.1
win-inet-pton          1.1.0
wincertstore           0.2
winrt                  1.0.21033.1
zipp                   3.5.0
```

dlech, 2022-01-25:
Uninstall the `winrt` package. It is no longer used and causes this error.

Novakasa, 2022-01-25:
That solved it, thanks.

closed 2022-01-25 by Novakasa