pybricks/support #620: [Question] Is the MoveHub integrated motor usable?

bestimmaa, 2022-01-28:
Since there is no mention in the API docs for the MoveHub - is it possible to use the integrated motor of the MoveHub?

laurensvalk, 2022-01-28:
Yes, you can use them like any other motor. They are labeled as A and B on the hub.

It is mentioned in a few places, but maybe we can make it clearer.

bestimmaa, 2022-01-28:
Thanks for your reply! If you don't mind: Where is this mentioned exactly? Maybe I overlooked this. I assumed it's a property of the hub class but the only features mentioned were lights, accelerometer and such.

dlech, 2022-01-28:
We should probably add it to this picture.

[image]

laurensvalk, 2022-01-28:
It's linked in the caption of the motor overview:

[image]

It doesn't really fit well in the picture since we need to show both sides.

dlech, 2022-01-28:
Looks OK to me.

[image]

bestimmaa, 2022-01-28:
Thanks for taking the time, I definitely missed this!

closed 2022-01-28 by bestimmaa