pybricks/support #624: [Doc]

estoykov, 2022-01-30:
**Describe the issue**
**Using the IMU** section is missing from Technic Hub documentation:
https://docs.pybricks.com/en/stable/hubs/technichub.html

**Suggested improvement**
IMU readings work just fine using BOOST hub examples, so nothing to code here, just needed to add the missing section in the Technic Hub page.

laurensvalk, 2022-01-30:
In the bottom left of the page you can choose the version of the docs.

If you select ‘latest’, you’ll see some documentation for the Technic Hub IMU as well.

It isn’t displayed in the stable version because there are still some bugs to be worked out.

I noticed you just became a Pybricks sponsor. Thank you!

estoykov, 2022-01-30:
Thank you for the clarification ;)
However choosing documentation version works only when browsing it in a separate tab.
I could not find similar feature in the right side documentation pane in code editor - both in browser or in beta PWA application.
Also, latest code editor (non-beta) is marked as v3.1 and when I select v3.1 in dedicated documentation page (https://docs.pybricks.com/en/v3.1.0/hubs/technichub.html), IMU section is there.
It seems that the code editors (both normal and beta) use "stable" version of the docs with missing IMU section (and probably more things).
So, if the current version of the code editor (non-beta - https://code.pybricks.com/) is v3.1 shouldn't the "stable" documentation to be v3.1 as well?
Thank you in advance and keep the good work!

laurensvalk, 2022-01-30:
Thanks for the additional feedback.

The beta is supposed to display the latest docs indeed, which is a bug tracked in https://github.com/pybricks/support/issues/612. This should get fixed when we release the next beta.

> Also, latest code editor (non-beta) is marked as v3.1 and when I select v3.1 in dediThanks for sharing this, this is a mistake as well. cated [documentation page](https://docs.pybricks.com/en/v3.1.0/hubs/technichub.html), IMU section is there.

This is a mistake as well and I hadn't seen it yet. Thanks for pointing this out.

laurensvalk, 2022-01-31:
Closing as this is being tracked in #612. Thanks!

closed 2022-01-31 by laurensvalk