pybricks/support #625: Is it possible to add some sort of "yeild" finction so I can download new code onto brick while old-code is running?

badassloumd, 2022-02-02:
Hello,
   Thank you for developing this wonderful software. I am
using Pybricks for my FLL team to program their robot(s)
and they are learning allot and doing quite well. Thank you.

  We are using an EV-3 and we have code that is structured
as a big while(1) loop that runs forever while polling the EV3
buttons like this
 
while(1):
        x=get_ev3_butttons()
        if(x==1) 
              do_some_some_stuff()
        if(x==2) 
              do_some_other_some_stuff()

   The issue is that you have to go over to the EV3
and stop the program before we can download new
code to the robot. The robot will not allow VS code to
send new code to the robot while the "old code" is
running. Is there a function I can put in to the while(1)
loop that can "detect" that VS code wants to send a 
new program down and stop running the wile(1) loop?

   Thank you.

dlech, 2022-02-02:
How are you downloading the program? If you use the VS Code debug feature to download and run, there is a stop button in VS Code.

See https://pybricks.com/ev3-micropython/startrun.html#downloading-and-running-a-program.

badassloumd, 2022-02-02:
Ha!! you gotta be kidding me!!I totally missed that!!!
OK next juicy question? Can I add breakpoints /  and
can I "single-step / resume" the code ??????

dlech, 2022-02-02:
Once upon a time, we had the vscode python debugger working on EV3, but it was very slow. See https://github.com/ev3dev/ptvsd.

If the `interactiveTerminal` is set to true in `.vscode/launch.json` in your project, then you can add `raise KeyboardInterrupt` to your code as a "breakpoint" (with no resume). You can also use the "pause" button to interrupt the code wherever it currently is. Both of these work the same. It just drops to a REPL in the terminal pane in VS Code. You can use the REPL to inspect the state of the program when it was interrupted. Note that local variable cannot be inspected, only global ones.

dlech, 2022-02-02:
Another tip for EV3 programming with infinite loops. Always add a `wait()` so that the OS has time to do other things besides run your program. The EV3 runs Linux, so it is not like other MINDSTORMS robots that don't have an OS.

If you don't do this, the stop button won't seem like it stops the program right away.

```python
from pybricks.tools import wait

while True:
   # do stuff
   ...
    # give OS time
    wait(10)
```

badassloumd, 2022-02-03:
Yeah that’s definitely true of most things with an operating system. I allways  put a wait() in long loops. was surprised to see that the new style Rick just appears to be running bare metal micro python on an STM32. I don't think its running Linux. - kind of a step backward.

badassloumd, 2022-02-21:
Hi. I was hoping you could help me with another issue. I’m trying to use the system.brickname() function With my EV three Pybrics but I just can’t seem to get it to work. Can you show me an example of how to get the brick name into a variable? Thank you.

dlech, 2022-02-21:
I don't think the `system` attribute is implement on EV3. But you can get the name of the brick like this:

```python
with open("/etc/hostname") as f:
    name = f.read().strip()
```

closed 2022-07-18 by dlech