pybricks/support #626: [Question] Is there standard procedure to install pybricks firmware on Spike Essential (small) hub?

estoykov, 2022-02-03:
Today I acquired this cute little piece of hardware and because its functionallity is very limitted via official Spike App I want to use it with pybricks.
I see that it should be supported (https://github.com/pybricks/support/issues/439) and there is a firmware in the CI builds, but there is no word for it in https://github.com/pybricks/support/issues/591.
When I connect the small hub via USB and click 'Connect' in the above install page, I get this:
[image]

However, I'm not sure that I can proceed with installing.

Is there some official way to install the latest working (beta ofc) pybrics build?

Thanks and keep the good work!

laurensvalk, 2022-02-03:
If you try, the installation will check the hub type and safely stop before actually doing any updates :)

So the temporary install page is only suited for Robot Inventor and SPIKE Prime at the moment. We could add SPIKE Essential Hub here also if there's demand for it, but we were thinking to just wait until we add the proper installer to our main Pybricks Code app.

You could still install it via [Pybricksdev](https://github.com/pybricks/pybricksdev) if you don't mind installing some Python packages.

estoykov, 2022-02-03:
I already have pybricksdev installed:
[image]
and if everything is correct in the picture, i.e. pybricksdev and Python version, I could give it a try.

I still need some more info though:
- There are a lot of builds in https://github.com/pybricks/pybricks-micropython/actions but from which build you suggest to get the essential hub firmware artifact? (Direct link will be OK as well). Exact pybricksdev command will be nice too :)
- Also, I don't see EssentialHub type or similar in both public API pages and in https://github.com/pybricks/pybricks-api
I suppose that its functionality (except Display and LEFT/RIGHT buttons) should be the same as the PrimeHub. With only two ports of course.

Overall, I see (in the code) that there is a support for this hub and if you think that it will work (with a little help from you) I'm happy to test it ;)

estoykov, 2022-02-03:
Okay, I'll give it a try. One last question regarding recovering, though:
In the Prime/Inventor hub install page, FW recovery process is described as running just a simple script from the beta code editor.
However, there is a an additional page - https://dfu.pybricks.com (if something went wrong) which shows how to install some backed up original FW in DFU mode and update it later via the original Lego software.
So, if there is possibility to do this with the small hub as well? I see that pybricksdev has an ability to save/restore firmware in DFU mode and I was able to put the hub in this mode (as the Windows reported) while holding the button and pluging in the USB (it flashes only in pink though).
Sorry, for bothering you again, but I want to be prepared as much as I can ;)

Regarding documentation, I see in the pybricks-micropython code that the class name (as I supposed) is EssentialHub ;)

dlech, 2022-02-03:
Yes, the `pybricks dfu restore` command that you found does the same as dfu.pybricks.com. The light won't change color until you actually run the command.

laurensvalk, 2022-02-03:
> So, if there is possibility to do this with the small hub as well? I see that pybricksdev has an ability to save/restore firmware in DFU mode and I was able to put the hub in this mode (as the Windows reported) while holding the button and pluging in the USB (it flashes only in pink though).

Yes, this aspect also works the same across these hub types.

We should probably add a downloadable backup file for the Essential Hub too, just in case people forget to make a backup.

When you install Pybricks `via pybricksdev flash path/to/essentialhub/firmware.zip` when the hub is in normal mode, it will store the backup on the hub itself.

All of this complexity (multiple ways to do the same thing) should be abstracted away eventually in Pybricks Code.

estoykov, 2022-02-03:
I did it ;)
1. Made a backup (after installing WinUSB for DFU mode as described in https://dfu.pybricks.com/extra.html)
[image]
2. Installed latest master build
[image]
3. And the success story:
[image]

Thank you all, you're the best!

P.S. The only thing that I'm really missing is to work from VS Code. No offense - your web code editor is REALLY great, but its intellisense is somewhat limited.

laurensvalk, 2022-02-18:
Somehow I missed your response. This is great, glad you got it working!

Meanwhile our [dfu page](https://dfu.pybricks.com/) has a backup file for the Essential hub too, just in case.

laurensvalk, 2022-02-18:
The flash size is 1 MB, so your backup also includes all of the empty space after it. Both will work.

And yes, you can set the name by adding `--name your_name` to the pybricksdev flash command.

Kefs88, 2022-03-03:
Looking good! My hopes are up that the Essential hub will get an official introduction together with the RI and Spike Hub :)
I know it's still new but having such a small unit to program will be so nice 🗡️ 

Keep up the good work!
(This seems a bit to hard for me to get working for now)

TheVinhLuong102, 2022-06-08:
I have just tried flashing the Essential Hub firmware by `pybricksdev flash essentialhub-firmware.zip` but encountered `KeyError: "There is no item named 'main.py' in the archive"`.

The firmware file was from the latest successful master build https://github.com/pybricks/pybricks-micropython/actions/runs/2457715521.

`pybricksdev` version is v1.0.0a25.

May I ask where I might have done it wrong?

Thank you David & Laurens!

laurensvalk, 2022-06-08:
This is something we're currently actively working on, so you may be better off using the latest beta release for the moment (instead of the *very* latest master build). You can download it here:
https://github.com/pybricks/pybricks-micropython/actions/runs/2436129709

If you are feeling adventurous, upgrade pybricksdev to the latest version and use the latest master firmware build. :smile:

TheVinhLuong102, 2022-06-08:
Laurens, I've just tried the adventurous route (`pybricksdev` a26 and the latest `master` build), and encountered the following error: :)

```
 File ".../pybricksdev", line 8, in <module>
    sys.exit(main())
  File ".../site-packages/pybricksdev/cli/__init__.py", line 376, in main
    asyncio.run(subparsers.choices[args.tool].tool.run(args))
  File ".../site-packages/pybricksdev/cli/__init__.py", line 224, in run
    from .flash import flash_firmware
  File ".../site-packages/pybricksdev/cli/flash.py", line 41, in <module>
    from ..flash import BootloaderConnection, create_firmware
  File ".../site-packages/pybricksdev/flash.py", line 25, in <module>
    from .firmware import ExtendedFirmwareMetadata, AnyFirmwareMetadata
  File ".../site-packages/pybricksdev/firmware.py", line 17, in <module>
    "device-id": Literal[0x40] | Literal[0x41] | Literal[0x80] | Literal[0x81],
TypeError: unsupported operand type(s) for |: '_LiteralGenericAlias' and '_LiteralGenericAlias'
```

dlech, 2022-06-08:
We change change that line to allow older Python for a while longer.

dlech, 2022-06-08:
You can apply this change locally and it will be fixed in the next release. https://github.com/pybricks/pybricksdev/commit/4060d1fc1561dacd879bb6e8c721e093a025572d

TheVinhLuong102, 2022-06-08:
David, I believe line 17 in that `firmware.py` module needs a change as well.

dlech, 2022-06-08:
yes, I missed that, thanks

dlech, 2022-06-08:
What operating system?

TheVinhLuong102, 2022-06-08:
I guessed so too, and have done those `brew install`s.

Now we're up against this next: :D 
```
pybricksdev flash essentialhub-firmware.zip
Creating firmware...
Connecting to /dev/cu.usbmodem3189376A34391
Connected!
Hub is ready.
Uploading _firmware/install_pybricks.py (8241 bytes)
Uploading _firmware/firmware.metadata.json (434 bytes)
Uploading _firmware/firmware.bin (231936 bytes)
Installing firmware

Detected LEGO firmware:
    Version: v1.0.00.0071-191f3ad
    Size: 584.0 KB
The detected firmware is not compatible with this installer.
Pybricks will not be installed.
```

dlech, 2022-06-08:
We haven't tested that firmware yet. You can edit `install_pybricks.py` inside of the `firmware.zip` file. That version will need to be added to this list:

```python
KNOWN_SPIKE_ESSENTIAL_FW_VERSIONS = {
    "v1.0.00.0000-d94a557": "SPIKE App v2.0.0",
    "v1.0.00.0070-51a2ff4": "SPIKE App v2.0.4",
}
```

TheVinhLuong102, 2022-06-08:
David, there's a problem installing `pybricksdev` in the editable mode: https://github.com/pybricks/pybricksdev/issues/32. Could you take a look at that as well?

dlech, 2022-06-08:
I already replied there.

dlech, 2022-06-08:
To ignore the problem and go a different direction, you could put the hub in DFU mode and flash it that way (turn off the hub and unplug USB, then press and hold the button until it flashes purple and plugin in the USB). Then run `pybricksdev flash` while the hub is in this state.

TheVinhLuong102, 2022-06-08:
David, yes, that now succeeds! :) 

Lots of hoops to jump through, but the most important thing is it now works.

Really looking forward to the new installer utility through Pybricks Code. Thank you for the awesome work!

```
pybricksdev flash essentialhub-firmware.zip
Creating firmware...
Could not find hub in standard firmware mode. Trying DFU.
File: /var/folders/_r/pclm3n6x2b54y_1frhn2jh3c0000gn/T/tmpinjct2hc/firmware.dfu
    b'DfuSe' v1, image size: 232229, targets: 1
    b'Target' 0, alt setting: 0, name: "ST...", size: 231944, elements: 1
      0, address: 0x08008000, size: 231936
    usb: 0694:000c, device: 0x0000, dfu: 0x011a, b'UFD', 16, 0x5c5d2dbb
Erasing flash...
Writing new firmware...
0x08008000  231936 [=========================] 100% 
Done.
```

laurensvalk, 2022-06-08:
Maybe I shouldn't have mentioned the adventurous approach :smile:

laurensvalk, 2022-06-08:
For what it's worth, you were very close. `"v1.0.00.0071-191f3ad"` does work just fine. We can add it to the list of known firmwares. And in your previous attempt, the file upload did work. Just not at the same time for some reason.

Once the easy install is included in Pybricks Code, I think we should drop this particular USB serial installer from pybricksdev and just use DFU only. That's one less thing to support/debug and no great loss.

dlech, 2022-07-28:
The new standard is to use https://beta.pybricks.com. Support for this hub was added in v3.2.0b3 (Pybricks Beta v2.0.0-beta.4).

closed 2022-07-28 by dlech