pybricks/support #634: [Bug] Firmware update stuck in "preparing the hub"

Cervesi, 2022-03-11:
I encountered this issue while trying to update the inventor hub firmware at https://install.pybricks.com/

I followed the steps until hitting install, after which the process stops at "preparing the hub" and does not make any progress.

[image]

Any idea of what that could be? Before this happened:
-I received the message from the main app that a new firmware was available
-I went to https://install.pybricks.com/, started the process, it stuck to "preparing the hub"
-I thought that it was because I did not update the official firmware for a while, so restored the hub using https://dfu.pybricks.com/ and then updated official firmware from Mindstorms app
-Tried https://install.pybricks.com/ again, but issue still present

laurensvalk, 2022-03-12:
Since you've used this before, is there any chance you are still seeing an outdated installation page?

It's hard to tell from the small screenshot. 

Could you try a hard refresh (ctrl + F5) or open it in a private window so we know the cache is cleared?

Cervesi, 2022-03-12:
Already cleared the cache and opened in private window, with no effect. The only information I have that something is wrong is the "preparing the hub message" at the end of the install page. I followed all the simple instructions that you provided.

The hub works just fine with the official app and firmware. Could it be that the update is just slow? Although I waited 30 mins and still nothing happened.

[image]

laurensvalk, 2022-03-12:
In that case it's probably a bug on our end. I'll look into it.

But since you're comfortable with using DFU mode for installation, you can also just install pybricks that way.

To find a recent build, see these instructions: https://pybricks.com/install/technic-boost-city/#installing-the-latest-build-advanced

But instead of following the final steps to install it, grab the  .bin file from the ZIP archive instead, and install it via the DFU installer.

laurensvalk, 2022-03-12:
Meanwhile, can you check the version of the mindstorms app that you used to install the latest official firmware? I'll see if I can reproduce it.

Cervesi, 2022-03-12:
Many thanks, I reinstalled the latest build using DFU and it is now working again. App version was 10.3.1.

closed 2022-06-30 by dlech