pybricks/support #636: [Question] Controlling multiple lego trains

JustStevee, 2022-03-20:
**Question**
Can I control multiple Lego trains at the same time?

JustStevee, 2022-03-21:
I want to make one program that will control multiple trains

BatteryPoweredBricks, 2022-07-28:
Very much interested in this as well.  My use case is a passenger train with powered locomotives at each end controlled by a single remote.  I'm currently using official firmware to accomplish this but the motor in one of the locomotives has to be reversed physically (or electrically).  I'd love to be able to use Pybricks to control both hubs simultaneously.

laurensvalk, 2022-07-29:
In the long run, this may be possible with https://github.com/pybricks/pybricks-micropython/pull/80

In this video, one hub is connected to the remote, and that hub broadcasts a signal to the others. In this example, each hub sets the same light color in response to a button press.

As you can see, it's not very fast, but probably good enough to set train speeds.

https://user-images.githubusercontent.com/12326241/181698251-fcd33351-c9c1-4197-a6ee-3a37b8ab34d0.mp4

dlech, 2023-02-24:
I'm going to go ahead and close this one since it is not technically possible to connect the LEGO Powered Up Remote to two other devices at the same time and we have other issues open already for alternate solutions.

closed 2023-02-24 by dlech