pybricks/support #638: [Doc] control.limits() not complete for EV3

XilefTech, 2022-03-24:
**Describe the issue**
https://pybricks.com/ev3-micropython/motors.html#pybricks._common.Control.limits
doesn't mention anycases in which the function couldn't be executed, yet when I try to call it while the motors run, I get an error that looks kinda like this:
```
OSError: [Errno 1] EPERM: 

The requested operation is not valid in the current state:
--> Check the documentation for required conditions.
--> Check the line in your script that matches
    the line number given in the 'Traceback' above.
```

**Suggested improvement**
mention that there are some circumstances under which you cannot change limits and maybe add, what these circumstances exactly are

laurensvalk, 2022-03-24:
> mention that there are some circumstances under which you cannot change limits and maybe add, what these circumstances exactly are

Just scroll up a little:
[image]

closed 2022-03-24 by XilefTech

laurensvalk, 2022-03-24:
No problem! The docs could probably be a bit clearer.

We are removing this restriction on newer releases (> 3.1), but there is currently not a specific plan to bring this to EV3.

XilefTech, 2022-03-24:
I think EV3 is still relevant as a plattform as many users still want to use their EV3 hardware. Would be a shame if there was no current software support/updates anymore

laurensvalk, 2022-03-24:
I agree, though it would be nice to know if this sentiment is shared more broadly.

If enough users want it, it could make sense to focus some effort on EV3. But if most people are happy with 2.0 the way it is, our effort might be better spent on the newer hubs.