pybricks/support #639: [Bug] Powered PUP sensors do not work

laurensvalk, 2022-03-24:
**Describe the bug**

The color sensor and ultrasonic sensor (and probably color matrix) lights are not working. This means the color sensor does not work at all.

This broke in https://github.com/pybricks/pybricks-micropython/commit/c66cadd23297aa647b54e27f757c3937f46e423c.

It looks like a `FIXME` was added concerning this, but it would be nice to keep the sensors working even if that means that the code doesn't look great :smile:

dlech, 2022-03-24:
So this just affects sensors that require extra power on pin 1/2?

laurensvalk, 2022-03-24:
I haven't tested any other sensors yet.

But UART motors work fine which usually means all 3v3 devices are working.

closed 2022-03-25 by dlech

laurensvalk, 2022-03-25:
For future reference (in case we ever go back in time to debug something else), I just realized this also fixes one port being always powered when a Color Sensor was plugged in. On Prime Hub, motor F would turn on automatically when plugged in. That is fixed now.