pybricks/support #642: Line 13 Errno 116

K2soo, 2022-04-07:
Hello

I have one question. What do I wrong.
I can't instal program to Lego 42124. 

After pairing the program with the hub, it blinks blue or turns solid blue.
After hitting the play button, it installs the code up to 95% and stops. I cannot complete the installation.
I install on the phone. Errors / locks are popping up on the computer when connecting the computer to the vehicle.

I am not an IT specialist. I just wanted to make a nice surprise for my son :/

Please help me

[image]

dlech, 2022-04-07:
> After hitting the play button, it installs the code up to 95% and stops.

what operating system is this on?

> Errors / locks are popping up on the computer when connecting the computer to the vehicle.

If this is the error in the screenshot, I'm guessing that the remote control was not turned on and there was a timeout waiting for the remote control to connect, but we can't see line 13 of the code where the error occurred, so we can't say for sure.

K2soo, 2022-04-07:
Thank you for your answer.
I turned on the remote. The LEDs in the car and the remote control are both blue. The wheels in the car turn left and right once - then nothing happens. The program does not stop. When after a few minutes I pressed the green button on the remote control, the installation ends at step 28.
I have attached a photo
-------- Oryginalna wiadomość --------Od: David Lechner ***@***.***> Data: 07.04.2022  18:40  (GMT+01:00) Do: pybricks/support ***@***.***> DW: K2soo ***@***.***>, Author ***@***.***> Temat: Re: [pybricks/support] Line 13 Errno 116 (Issue #642) 

After hitting the play button, it installs the code up to 95% and stops.

what operating system is this on?

Errors / locks are popping up on the computer when connecting the computer to the vehicle.

If this is the error in the screenshot, I'm guessing that the remote control was not turned on and there was a timeout waiting for the remote control to connect, but we can't see line 13 of the code where the error occurred, so we can't say for sure.

laurensvalk, 2022-04-07:
Are you running this program?

https://pybricks.com/projects/sets/technic/42124-off-road-buggy/powered-up-remote/

K2soo, 2022-04-07:
Yes, I uploaded the code through Your website.
I turned on the remote control. The LEDs in the car and on the remote control are blue. The wheels turn left and right - then nothing happens. The program does not stop. When after a few minutes I pressed the green button on the remote control, the installation ends at step 28.
I'm sending a photo.

I will ask more. After trying to install the code on Your part, I cannot connect to the LEGO program via the remote control or by phone. As far as I remember, the white LED was blinking after turning on the hub. Currently it only blinks blue.
Do you have a code that was uploaded by the LEGO manufacturer?
-------- Oryginalna wiadomość --------Od: laurensvalk ***@***.***> Data: 07.04.2022  19:50  (GMT+01:00) Do: pybricks/support ***@***.***> DW: K2soo ***@***.***>, Author ***@***.***> Temat: Re: [pybricks/support] Line 13 Errno 116 (Issue #642) 
Are you running this program?
https://pybricks.com/projects/sets/technic/42124-off-road-buggy/powered-up-remote/

K2soo, 2022-04-07:
I'm sending a photo
-------- Oryginalna wiadomość --------Od: laurensvalk ***@***.***> Data: 07.04.2022  19:50  (GMT+01:00) Do: pybricks/support ***@***.***> DW: K2soo ***@***.***>, Author ***@***.***> Temat: Re: [pybricks/support] Line 13 Errno 116 (Issue #642) 
Are you running this program?
https://pybricks.com/projects/sets/technic/42124-off-road-buggy/powered-up-remote/

dlech, 2022-04-07:
GitHub does not allow attachments via email. You have to use the website.

K2soo, 2022-04-07:
[image]

K2soo, 2022-04-09:
I'm still asking for help.
 I stopped on line 28 in code. I installed everything from scratch. After switching on the code upload, the remote control lights up, and the blue LED flashes on the car. The car turns itself left and right with its wheels. Still nothing happens. When I press the button after a long moment, it understandably throws an error in code 28.
I feel that I am close to the end of the installation.

Please help!

I carry out the entire installation on an android phone on chrome.
On pc, it crashes standard web bluetooth issue on every browser.

[image]

laurensvalk, 2022-04-09:
When you copied the code, it looks like not all spaces at the start of the lines were included. You'll want to make sure that the code looks exactly the same as on the page you copied it from.

That's why it's usually easier to do this on a PC. Which operating system are you using? Windows or Mac? Which version?

And which browser did you try. You'll need Google Chrome to try this.

K2soo, 2022-04-09:
I checked the code from the computer with the phone. I deleted unnecessary spaces.
It works!! haha

Thank you for your help!!

Finally, I will ask.
Do I have to upload the code from the tablet every time I want to connect the remote control to the car?

dlech, 2022-04-09:
You can check this button and flash the firmware again to save the program to the hub.

[image]