pybricks/support #644: [Question] Is it available to use Bluetooth UART function in V3.2?

jasonxfeng, 2022-04-20:
**Question**
Is it available to use Bluetooth UART function in V3.2?

**Context**
I am the new Joiner of pybricks. I flashed V3.2 by Automatic firmware updates. I am trying to control the hub by my own app via Bluetooth but I can't import pybricks.ble , ubluetooth and micropython as well . Any ideas to make it work in V3.2?

Thank you so so much! <3

**Screenshots**
[image]

dlech, 2022-04-21:
Currently, `sys.stdin` and `sys.stdout` use the nRF UART service.

jasonxfeng, 2022-04-21:
Thanks Dlech, So it has not bluetooth communication function in V3.2, does it?

dlech, 2022-04-21:
Technical it does have "Bluetooth communication". It just may or may not be useful depending on what exactly you want to do with it.

jasonxfeng, 2022-04-22:
Hi dlech,

I wanna do "Other BLE UART devices"
I am using inventor and i dont know which firmware can support this function...

[image]

laurensvalk, 2022-05-02:
If you need something that works with today's code, you can try this: https://github.com/pybricks/support/issues/470#issuecomment-1016576294

jasonxfeng, 2022-05-03:
HI,
I solved the question already. I am using Inventor Hub.
I just wanna call the BLE function and i tried to flash pybricks FW v3.1 into HUB.
But It shows error when I tried to import BLE.
Finally, I found out the officail FW can import BLE function from micropython.

Thank you very much~ ;)

closed 2022-05-03 by jasonxfeng

jononi, 2022-05-19:
> 

Hi @jasonxfeng ,
can you share some more info on the solution you went with? 
My use case is a BLE sensor (ESP32 based) that continuously sends info over nRF UART service. I want to read the data on the Inventor hub while also maintaining communication with code.pybricks.com. Is that even possible?

dlech, 2022-05-19:
> . I want to read the data on the Inventor hub while also maintaining communication with code.pybricks.com. Is that even possible?

Not currently. Right now, there is only one Nordic UART Service instance on the hub, so it could only be used for one or the other, but not both.

jasonxfeng, 2022-05-19:
> read the data on the Inventor hub

Hi @jononi, my use case is controling the Inventor hub to drive by receving data from other STM32 device with BLE module. I dont know is it the same use case with you.

laurensvalk, 2022-10-25:
There is now a tutorial here: https://pybricks.com/projects/tutorials/wireless/hub-to-device/pc-communication/

[image]

So we can close this issue.