pybricks/support #646: [Bug] Hub stays connected when closing Pybricks Code in Ubuntu 22.04

laurensvalk, 2022-04-23:
**Describe the bug**
The hub stays connected when closing Pybricks Code in Ubuntu 22.04.

**To reproduce**
Steps to reproduce the behavior:
1. Connect to the hub (tested with Technic Hub)
2. Close Pybricks Code app window or browser

Hub LED stays solid white. It also stays connected in the Ubuntu UI:

Disconnecting from there does work.

[image]

**Expected behavior**
Disconnect on closing the browser.
@

laurensvalk, 2022-04-29:
This also happens when clicking the disconnect button.

dlech, 2022-06-03:
logs:
[image]

dlech, 2022-06-03:
Upstream bug: https://crbug.com/1332640

dlech, 2022-06-28:
This should be fixed in chromium v104.

dlech, 2022-08-05:
Confirmed fixed in chromium 104.

closed 2022-08-05 by dlech

laurensvalk, 2022-11-22:
Is this related to disconnecting being quite slow now or is that a separate issue? On Ubuntu it can take up to 5 seconds now for the disconnect button to complete.

dlech, 2022-11-22:
> Is this related to disconnecting being quite slow now or is that a separate issue? On Ubuntu it can take up to 5 seconds now for the disconnect button to complete.

The 5 second delay is intentional as workaround for a different upstream bug. https://github.com/pybricks/pybricks-code/commit/509c635a07d1267d9417601f3929c5675befe080