pybricks/support #647: Charging an Inventor Hub via MacBook never finishes

lobodpav, 2022-04-25:


laurensvalk, 2022-06-17:
I am seeing values on Prime Hub with Pybricks similar to @lobodpav.

In my case, the hub is connected to the USB port on my monitor, so this issue might be fairly common.

dlech, 2022-08-10:
As mentioned in duplicate issue #700, I've now seen this when leaving a hub running overnight. The hub had 5 motors plugged in (not running) and a SPIKE color sensor with the lights on along with the normal status lights on the hub itself contributing to the 90 mA load. I logged the data during the time and graphed it:

[image]

We can see that we are in the constant-voltage phase of the charging cycle as previously thought. In this phase, the current is supposed to drop to below 150mA and then charging stops. But we can see the current never drops below 300mA so charging doesn't stop and the safety timer eventually kicks in and the battery starts discharging.

dlech, 2022-08-10:
The jump in the charge current in the graph could be explained by this from the data sheet: "When the input current at the VL side is lower than 330mA, the MP2639A turns off Q1 and switches to non-sync operation.". Unfortunately, it doesn't go into any more detail about what "non-sync operation" is. Nevertheless, seems like a reasonable suspect for why the charge current would stall or reach equilibrium at this particular level.

laurensvalk, 2022-08-10:
Could you add the charger type to your log post? Does it only happen on standard downstream ports?

dlech, 2022-08-10:
That information is not exposed in the Python API so we have no way of knowing what type of charger the firmware has detected. But assuming it was correctly detected, it is a USB 2.0 port, so should be detected as standard downstream.

laurensvalk, 2022-08-10:
It looks like the battery current is always 90 mA for this load, whether it not it is charging. Does it remain 90 mA when unplugged?

If so, where is that 300 mA going all this time? It would be interesting to see if we could measure the real current through the cable for comparison.

dlech, 2022-08-10:
I actually just did that while you were typing :wink:. I've been running another experiment an the charge current measured by the hub is 335mA. The current measured by my USB power meter is 0.21A. However, the data sheet does state that what we call the "charge current" is the total battery current, not what is just coming in over USB.

laurensvalk, 2022-08-10:
Looking forward to seeing that experiment when it ends, and whether that 0.21A drops once the battery is full.

dlech, 2022-08-10:
I have repeated the experiment, this time with nothing attached to the I/O ports on the hub and all of the lights turned off to simulate the hub being "shut down". This time the battery current measured on the hub is about 15mA instead of 90mA. Also this time, it is starting with a not full battery so we can see the constant current charge phase before the constant voltage charge phase.

[image]

We can see that the "charge current" still levels off, this time at around 250mA. The USB power meter reads 0.07A at the end.

Comparing this to a similar graph from the data sheet, it seems like the current should be dropping much faster instead of leveling.

[image]

dlech, 2022-08-10:
More observations:

When the battery is "full" (8.348V for this hub) and I unplug the USB and plugin it back in, I get different results depending on the port. When plugging into my computer, the battery indicator light shows charging (red) and the meter shows the port is at 4.88V. If I plug into a wall socket USB charger the battery status light on the hub show charge complete (green) and the USB meter measures 5.13V. The data sheet mentions that the charge current is also proportionally limited by the USB voltage, so this could be a factor as well.

dlech, 2022-08-11:
Another experiment, same hub, all lights off, no I/O devices, connected to USB charging port (wall socket type) so it charges with a max of 1.5A instead of 500mA. Also supplies voltage a bit over 5V as mentioned above. 

This time, the graph looks more like the one from the data sheet. The charge current actually starts dropping proportionally during the "constant current" phase then drops much more quickly during the "constant voltage" charge phase. Then we get CHGOK and the battery starts discharging.

[image]

dlech, 2022-08-11:
Raw data for the record:

[RI hub battery charging logs 2022-08-10.ods](https://github.com/pybricks/support/files/9310431/RI.hub.battery.charging.logs.2022-08-10.ods)

closed 2022-08-12 by dlech

laurensvalk, 2022-08-12:
> [dlech](https://github.com/dlech) closed this as [completed](https://github.com/pybricks/support/issues?q=is%3Aissue+is%3Aclosed+archived%3Afalse+reason%3Acompleted) in https://github.com/pybricks/pybricks-micropython/commit/34ce9817611f4b5617b4fbb2d3f51eac61a662ef [28 minutes ago](https://github.com/pybricks/support/issues/647#event-7180871886)

Besides the improved light indication, does this also stop charging in the case where it never finishes above? (In order protect the battery by mimicing the same discharge behavior of the wall charger?)

dlech, 2022-08-12:
No, that has not been addressed.