pybricks/support #648: [Question] Any way to run version 3.0+ on the EV3?

Nawor3565, 2022-04-26:
**Question**
I can't seem to find a clear answer as to whether there's any version of Pybricks past version 2.0.0 that works with the Mindstorm EV3. It seems like SD card images only exist for V2, so I would assume that support for the EV3 was dropped passed that release, but I would like to make sure.

**Context**
I tried to use the driver.curve() function on my EV3, only to get an error `AttributeError: 'DriveBase' object has no attribute 'curve'`. It appears that the function was added at some point in version 3, and since it would make my code far easier to write, I was hoping I could update the Pybricks software that came with my Mindstorms brick.

dlech, 2022-04-26:
Please see https://github.com/pybricks/support/discussions/370#discussioncomment-885240 for the answer.

Nawor3565, 2022-04-29:
Thanks! I tried what the comment said but it didnt seem to work. I tried placing the `pybricks-micropython` file in the specified directory both on the micro-SD card and on the Mindstorm itself via SSH/SFTP (rebooting afterwards both times), but the curve() function still doesn't exist so I assume the update didn't work. Any idea what I might be doing wrong?

Lemru, 2022-05-05:
The [link](https://github.com/pybricks/pybricks-micropython/suites/3003726393/artifacts/67940441) in [#370 (comment)](https://github.com/pybricks/support/discussions/370#discussioncomment-885240) doesn't seem to be available anymore. Where could I download the latest build instead?

Thanks in advance for your support!

dlech, 2022-05-05:
This link will always give the "latest" builds: https://github.com/pybricks/pybricks-micropython/actions/workflows/build.yml?query=is%3Asuccess+branch%3Amaster

Click on the first one in the list the scroll down to *Artifacts* and click `pybricks-micropython` to download the file.

Lemru, 2022-05-07:
Found it! Thanks for the explanation. Would it be necessary to update the files in /usr/lib/pybricks-micropython/ as well? If so, where would I find them?

dlech, 2022-05-07:
> Would it be necessary to update the files in /usr/lib/pybricks-micropython/ as well?

No, there is nothing to update here.

Lemru, 2022-05-08:
Thanks for spending your time on this. I raised the question, because I get following error message with pybricks-micropython 3 build 1733:

    robot@ev3dev:~/StallTest$ brickrun -r pybricks-micropython main.py
    Traceback (most recent call last):
      File "main.py", line 7, in <module>
    ValueError: incompatible .mpy file

Following the code of the test application, which worked well with version 2 - the one, which was installed by default. I only adopted the number and values of some function parameters accordingly:

    #!/usr/bin/env pybricks-micropython
    from pybricks.hubs import EV3Brick
    from pybricks.ev3devices import Motor
    from pybricks.parameters import Port, Direction
    from pybricks.tools import wait, DataLog, StopWatch
    
    from threading import Thread
    
    
    ev3 = EV3Brick()
    middle_motor = Motor(Port.B, Direction.COUNTERCLOCKWISE)
    
    def logging(log : bool):
        data = DataLog(' time, angle, speed', name='data', timestamp=False)
        watch = StopWatch()
    
        while log():
            data.log(
                '{:.3f}'.format(watch.time() / 1000),
                '{:5d}'.format(middle_motor.angle()),
                '{:5d}'.format(middle_motor.speed())
            )
            wait(10)
        
    if __name__ == '__main__':
        log_enabled = True
        t = Thread(target = logging, args=[lambda: log_enabled])
        t.start()
    
        # limits defaults: 1000, 4000, 200
        middle_motor.control.limits(10000, 10000, 200)
        # pid defaults: 3000, 10002, 30, None, 5
        middle_motor.control.pid(3000, 10002, 30, None, 5)
        # stall_tolerances defaults: 15, 200
        middle_motor.control.stall_tolerances(100, 20)
    
        # move lever to the left stop:
        middle_motor.run_until_stalled(-2000)
        middle_motor.reset_angle(0)
        # move lever to the right stop:
        max_angle = middle_motor.run_until_stalled(2000)
    
        log_enabled = False

There seems to be an issue with threading. If I remove the thread related lines, the script is working again. Do you have an idea, how to fix this?

dlech, 2022-05-08:
> from threading import Thread

I didn't think that the `.mpy` version changed since Pybricks v2.0, but maybe it did. I would suggest avoiding threads if possible, but if not, import from `_thread` instead of `threading`.

Nawor3565, 2022-05-08:
@Lemru, would you be able to tell me what you did to get 3.1.0 to run? Did you put the updated file on the SD card or the Mindstorm itself? And was that all you did?

Lemru, 2022-05-09:
@Nawor3565: 

Start your EV3 brick with an EV3DEV SD card inserted. Establish a network connection and look on your brick for the IP address. If you have never done this before, this [Tutorials](https://www.ev3dev.org/docs/tutorials/) page might help you. Connect to your brick via ssh:

    ssh robot@<ip-address of your EV3 brick>

You will be asked for a password. The default one is 'maker'.
Switch to the folder /usr/bin with root rights:

    cd /usr/bin
    sudo -s

The password you are asked for is 'maker' again.
Make a backup of the installed pybricks-micropython module:

    mv pybricks-micropython pybricks-micropython_2.0.0

Now you can copy the new module from the downloaded zip file to your brick.
On Linux you open a terminal, on Windows a command shell or a PowerShell, and switch to the directory with the extracted pybricks-micropython module. Then you type following command:

    scp pybricks-micropython robot@<ip-address>:/usr/bin/

and enter the password as requested. Now you need to make the new module executable. Go back to your ssh shell and to the directory /usr/bin with root rights. If you enter

    ls -la pybricks-micropython

you should see one line starting with "-rw-r--r-- 1 root root", which indicates that pybricks-micropython is not yet executable. You need to change this with

    chmod ugo+x pybricks-micropython

If you enter again

    ls -la pybricks-micropython

the first characters of the line should have changed to "-rwxr-xr-x 1 root root"

You are done!

Nawor3565, 2022-05-09:
Thank you so much! I didn't realize I needed to chmod the file, I'll try that today!

Lemru, 2022-05-10:
You are right, I updated the instruction in case someone should stumble accross this issue.

Why didn' version 3 ever make it into the default EV3DEV firmware? What needs to be done to make it happen?

dlech, 2022-05-10:
We've been focused on adding support for newer hubs (an there are lots of them!) for v3.x.

Lemru, 2022-05-10:
I thought so. I'm coaching WRO Teams in my spare time and we are still using EV3 bricks, at least as long as they work properly. I know from my children as well as from other coaches that many schools are in the same situation. They bought lots of EV3 Education packs and get more and more the impression that Lego has let them down. There is neither the need nor the money to move over to the new spike equipment, at least not as long as the EV3 is competitive. Because of Corona there were no competitions here last year, in 2020 they were canceled. We are using VS Code and started late this year. So we only recently realized that there were some inexplicable differences in the code completion and the pybricks documentation. Investigating these issues led me finally to this discussion. I appreciate your work very much, but I was a bit disappointed to see that there were no pybricks updates since 2020 for the EV3, while the spike hub was obviously more in focus. My concern is to be able to continue using the EV3 without restrictions in competitions such as WRO and FLL. So my first approach was to get the differences in VS Code sorted out. But my overall concern is, to keep the EV3 up to date as long as possible. To my opinion the EV3 still has significant advatages over the spike hub with two additional ports.

Therefore my question: How can I support you to keep the pybricks EV3 project alive?

dlech, 2022-05-10:
> So we only recently realized that there were some inexplicable differences in the code completion and the pybricks documentation

FYI, there are 2.x docs at https://pybricks.com/ev3-micropython/ and 2.x code completion can be installed with `pip install pybricks<3`. If something doesn't match between these, please open an issue and we will fix it.

We don't know of any major issues with 2.x so even though there haven't been any updates, it still should work well for most cases.

> Therefore my question: How can I support you to keep the pybricks EV3 project alive?

Ideally by finding some funding one way or another: https://github.com/sponsors/pybricks

This is of course more than could be expected of one individual or team but maybe someone like you could do the legwork to find a grant or promote some crowdsourcing with others that feel the same way about keeping EV3 alive?

dlech, 2023-05-26:
Update: there is now a new debian package for Pybricks v3.x on ev3dev, see https://github.com/pybricks/support/issues/1034#issuecomment-1564873392

dlech, 2025-07-13:
Closing this since we're dropping support for ev3dev in favor of a new EV3 solution.

closed 2025-07-13 by dlech