pybricks/support #655: About lowering amp to L-size motor to prevent choking of M-size motor

garfield8625, 2022-05-19:
### Discussed in https://github.com/pybricks/support/discussions/654

<div type='discussions-op-text'>

<sup>Originally posted by **garfield8625** May 19, 2022</sup>
Hello wise-ones,

I have the following issue:     While steering is engaged/actively used it chokes the driving motor
Question:     Is there a way to lower the amp that the Large motor gets via code to make it leave more juice for the M-sized?

Lego ECTO-1 motorized: 
- hub: 88009
- steering: 1 Large motor (88013), 
- drive: 1 Medium linear motor (45303)
(Medium motor is being used due to lack of space ofr bigger motor.)

# Code (if needed): 

initial_drive_speed = 100

steer = Motor(Port.B, Direction.COUNTERCLOCKWISE)
back = DCMotor(Port.A, Direction.COUNTERCLOCKWISE)

remote = Remote()

old_kp, old_ki, old_kd, _, _ = steer.control.pid()

steer.control.pid(kp=old_kp*4, kd=old_kd*0.4)

left_end = steer.run_until_stalled(-200, then=Stop.HOLD)
right_end = steer.run_until_stalled(200, then=Stop.HOLD)

steer.reset_angle((right_end - left_end)/2)
steer.run_target(speed=200, target_angle=0, wait=False)

steer_angle = (((right_end - left_end)/2)-5)
print('steer angle:',steer_angle)

while True:
    pressed = remote.buttons.pressed()

    if Button.RIGHT_PLUS in pressed:
        steer.run_target(1400, -steer_angle, Stop.HOLD, False)
    elif Button.RIGHT_MINUS in pressed:
        steer.run_target(1400, steer_angle, Stop.HOLD, False)
    else:
        steer.track_target(0)

    drive_speed = 0
    if Button.LEFT_PLUS in pressed:
        drive_speed = initial_drive_speed
    if Button.LEFT_MINUS in pressed:
        drive_speed = -initial_drive_speed

    back.dc(drive_speed)

    wait(10)

</div>

dlech, 2022-05-19:
There is a maximum voltage setting for motors: https://docs.pybricks.com/en/stable/pupdevices/motor.html#pybricks.pupdevices.Motor.settings

There is also a maximum torque setting for controlled actuation: https://docs.pybricks.com/en/stable/pupdevices/motor.html#pybricks.pupdevices.Motor.Motor.control.limits

dlech, 2022-05-19:
closing the issue since this is better suited to a discussion and there is a duplicate post there.

closed 2022-05-19 by dlech