pybricks/support #656: OSError: 112 when trying to connect via bluetooth.

MrCheatEugene, 2022-05-23:
Hi there. I am trying to recieve the message from EV3 with MicroPython, and send from EV3 with the stock OS.

But, I have trouble connecting them. While they are paired and connected, the program crashes with 'OSError: 112'.
Code: `
#!/usr/bin/env pybricks-micropython
from pybricks.hubs import EV3Brick
from pybricks.ev3devices import (Motor, TouchSensor, ColorSensor,
                                 InfraredSensor, UltrasonicSensor, GyroSensor)
from pybricks.parameters import Port, Stop, Direction, Button, Color
from pybricks.tools import wait, StopWatch, DataLog
from pybricks.robotics import DriveBase
from pybricks.media.ev3dev import SoundFile, ImageFile
from pybricks.messaging import BluetoothMailboxClient, NumericMailbox
ev3 = EV3Brick()

client = BluetoothMailboxClient()

mxox = NumericMailbox('mouse',client)
print('establishing connection...')
client.connect('BRICKNAME')
print('connected!')
mbox.wait()
print(mbox.read())
`

Full traceback:
`Starting: brickrun --directory="/home/robot/mouse" "/home/robot/mouse/main.py"
----------
establishing connection...
Traceback (most recent call last):
  File "/home/robot/mouse/main.py", line 22, in <module>
  File "pybricks/messaging.py", line 321, in connect
  File "pybricks/messaging.py", line 318, in connect
  File "pybricks/bluetooth.py", line 168, in handle_request
OSError: 112
----------
Exited with error code 1.
`
Why could that happen?

laurensvalk, 2022-05-23:
Thanks for sharing this code. This looks like the code from one of your EV3s. Can you share the code of the other one too?

What is the brick name of this EV3, and what is the name of the other?

MrCheatEugene, 2022-05-23:
> Thanks for sharing this code. This looks like the code from one of your EV3s. Can you share the code of the other one too?
> 
> What is the brick name of this EV3, and what is the name of the other?

The code of the second EV3 is lost, unfortunatly, basically it was a BT connection, then a loop that had a switch inside of it, which checked the state of which button on EV3 is pressed, and then it sent the numeric message with mailbox name 'mouse',
and with contents from 1 to 5.

Name of the stock firmware EV3 is EV31, and the name of the ev3 with micropython is ev3dev.

MrCheatEugene, 2022-05-23:
> The code of the second EV3 is lost, unfortunatly, basically it was a BT connection, then a loop that had a switch inside of it, which checked the state of which button on EV3 is pressed, and then it [...]

Also, the code on the second EV3 was not written in python, but in the LEGO EV3 Home edition software for Windows.

MrCheatEugene, 2022-05-23:
I've forgot one little thing. That program written using EV3's software tried to connect to EV3, not ev3dev.
Re-writing this program again, and after making some more changes to Python script, it now works perfectly.
The final Python code:
`#!/usr/bin/env pybricks-micropython
from pybricks.hubs import EV3Brick
from pybricks.ev3devices import (Motor, TouchSensor, ColorSensor,
                                 InfraredSensor, UltrasonicSensor, GyroSensor)
from pybricks.parameters import Port, Stop, Direction, Button, Color
from pybricks.tools import wait, StopWatch, DataLog
from pybricks.robotics import DriveBase
from pybricks.media.ev3dev import SoundFile, ImageFile
from pybricks.messaging import BluetoothMailboxClient, NumericMailbox
import time
# This program requires LEGO EV3 MicroPython v2.0 or higher.
# Click "Open user guide" on the EV3 extension tab for more information.

# Create your objects here.
ev3 = EV3Brick()

client = BluetoothMailboxClient()

mbox = NumericMailbox('mouse',client)
print('establishing connection...')
client.connect('MRCHEATGA')
print('connected!')
mbox.wait()
while(True):
    print(mbox.read())
    time.sleep(1)
    pass
# Write your program here.
#ev3.speaker.beep()
`
[The final EV3 code: mouse.zip](https://github.com/pybricks/support/files/8753896/mouse.zip).

closed 2022-05-23 by MrCheatEugene

laurensvalk, 2022-05-23:
Glad you figured it out.

>[The final EV3 code: mouse.zip](https://github.com/pybricks/support/files/8753896/mouse.zip).

If it's not too much trouble, would you mind pasting a screenshot of that program here? That would make it easy for people who find this issue via Google in the future.

MrCheatEugene, 2022-05-23:
> Glad you figured it out.
> 
> > [The final EV3 code: mouse.zip](https://github.com/pybricks/support/files/8753896/mouse.zip).
> 
> If it's not too much trouble, would you mind pasting a screenshot of that program here? That would make it easy for people who find this issue via Google in the future.

[image]
Here it is.