pybricks/support #661: Essential Hub does not run program (though syntax errors are correctly detected)

TheVinhLuong102, 2022-06-08:
Background: I've successfully installed the Pybricks firmware on the Essential Hub, following the discussion here:
https://github.com/pybricks/support/issues/626.

I'm running a simple program as follows, through both beta.pybricks.com and `pybricksdev run ble ...`:

```
from pybricks.hubs import EssentialHub
from pybricks.pupdevices import Motor, ColorSensor, ColorLightMatrix
from pybricks.parameters import Button, Color, Direction, Port, Stop
from pybricks.robotics import DriveBase
from pybricks.tools import wait, StopWatch

hub = EssentialHub()

matrix = ColorLightMatrix(Port.A)
motor = Motor(Port.B)

matrix.on(Color.GREEN)
motor.run_time(1000, 3, then=Stop.HOLD, wait=True)

```

Observations:
- Any syntax errors are correctly detected -- if I mess up the program with a few typos, the runner will yell
- However the correct program when run does not produce any observable actions by the robot components -- no green light, no motor movement

dlech, 2022-06-08:
This is telling the motor to run for 3 milliseconds, so it doesn't have time to start moving.

dlech, 2022-06-08:
I am getting the green lights on the light matrix when I run this program and the motor moves if I increase the time.

TheVinhLuong102, 2022-06-08:
oh yes, I forgot the time unit is milisecs, not seconds :D It totally works. Closing this issue now. :)

closed 2022-06-08 by TheVinhLuong102