pybricks/support #662: [Bug]  run_until_stalled() doesn't return correct angle value in 3.2.0b1

jononi, 2022-06-08:
**Describe the bug**
motor.run_until_stalled() used to block until motor is stalled then it returns the angle at which is stopped. With 3.2.0b1 it seems like it doesn't block anymore and it returns a wrong value, although the motor runs until it stalls .

**To reproduce**
```
(...)
>>> print(pybricks.version)
('technichub', '3.2.0b1', 'v3.2.0b1 on 2022-06-03')
steer = Motor(Port.C)
>>> steer.angle()
-102
>>> steer.run_until_stalled(300, duty_limit=30)
-102
steer.angle()
105
```

**Expected behavior**
run motor until it stalls, then return the angle value it stopped at.

laurensvalk, 2022-06-09:
Thanks for reporting! There's certainly something wrong here, even though the behavior I'm seeing seems to be different.

When I run this, the command does block and return the correct value upon stalling. But then it keeps going instead of stopping.

In your example snippet above, was there anything before this? Do you see the same behavior after a full reset and running only those commands?

laurensvalk, 2022-06-09:
This bug affects only the coast and brake stop type. As a workaround, you could use hold for now:

```python
from pybricks.pupdevices import Motor
from pybricks.parameters import Port, Stop
from pybricks.tools import wait

# Initialize a motor on port A.
example_motor = Motor(Port.A)

# Run at 500 deg/s until the motor stalls
angle = example_motor.run_until_stalled(500, duty_limit=30, then=Stop.HOLD)

print("Stopped at", angle)
```

laurensvalk, 2022-06-09:
This was broken in https://github.com/pybricks/pybricks-micropython/commit/4efeee25d39a339e141b7fa8fa919b36d0a6f74d#diff-9c5fdeeb615547288d3470c9ca282d080197673d1d9bb2c2800584a46f336d67L352

laurensvalk, 2022-06-09:
I can sometimes also reproduce your issue about returning right away, which looks to be separate from the issue mentioned above.

laurensvalk, 2022-06-09:
If you want to try out the fix on the Move Hub / City Hub / Technic Hub, use [these instructions](https://pybricks.com/install/technic-boost-city/#installing-the-latest-build-advanced) and install [this build](https://github.com/pybricks/pybricks-micropython/actions/runs/2466657896).

closed 2022-06-09 by laurensvalk