pybricks/support #665: [Bug] Reported current usage is very high

jononi, 2022-06-11:
**Describe the bug**

On Technic Hub running pybricks 3.2.0b1:

The battery current as reported by `hub.battery.current()` is too high and it looks like it's a reporting error because at that value the unit should be getting warm. I'm not noticing a faster discharge of the batteries. When connecting motors they run a bit slower like when batteries are not fully charged but nothing like when on almost empty batteries.

**To reproduce**
Connect Technic Hub with nothing attached. Run:
```
from pybricks.hubs import TechnicHub
hub = TechnicHub()
print('V(bat): {} mV'.format(hub.battery.voltage()))
print('I(bat): {} mA'.format(hub.battery.current()))
```
Output:
```
V(bat): 8667 mV
I(bat): 1381 mA
```

**Expected behavior**
current load in the ~80mA (when comparing to a unit with Lego firmware installed)

dlech, 2022-06-11:
Thanks for reporting. I am able to reproduce the issue. The battery voltage is reading too high as well.

dlech, 2022-06-11:
Reviewing the git history shows there was a major update in the STM32L4 HAL that was included when we updated to a newer version of MicroPython. No Pybricks code changed. So HAL library changes are likely the cause.

I hacked in a function to read the ADC registers to see what is different.

```python
adc, common = hub.battery.reg()
print('ADC')
print(*("{:02x}: {:02x}".format(i, x) for i, x in enumerate(adc)), sep='\n')
print('ADC_COMMON')
print(*("{:02x}: {:02x}".format(i, x) for i, x in enumerate(common)), sep='\n')
```

Which produces the following diff:

```diff
--- <v3.1.0>
+++ <v3.2.0b1>
@@ -63,7 +63,7 @@
 3d: 00
 3e: 00
 3f: 00
-40: b1
+40: b3
 41: 03
 42: 00
 43: 00
@@ -98,7 +98,7 @@
 60: 00
 61: 00
 62: 00
-63: 00
+63: c4
 64: 00
 65: 00
 66: 00
@@ -175,7 +175,7 @@
 ad: 00
 ae: 00
 af: 00
-b0: 00
+b0: 0c
 b1: 00
 b2: 00
 b3: 00
@@ -196,7 +196,7 @@
 09: 00
 0a: 83
 0b: 00
-0c: b1
+0c: b3
 0d: 03
 0e: 00
 0f: 00
```

- ADC 0x40 is data register (ADC_DR) so expected to be different most of the time as it is always changing as data is read.
- ADC 0x63 is part of  ADC_OFR1 - OFFSET1_EN << 7 + OFFSET1_CH << 2 - before not enabled - after enabled, offset of 1 - channel 1 is not used on the technic hub
- ADC 0xb0 is part of ADC_DIFSEL - lsb of DIFSEL - before no channels configured in diff mode - after ch 3 configured in diff mode - channel 3 is the battery current :raised_eyebrow: 
- ADC_COMMON 0x0c is part of ADC_CDR - lsb of RDATA_MST - this is a data register for dual mode

dlech, 2022-06-11:
The fix turned out to be trivial. After the fix:

```
('technichub', '3.2.0b1', 'v3.2.0b1-10-g4fc6f9d2-dirty on 2022-06-11')
V(bat): 7546 mV
I(bat): 135 mA
```

closed 2022-06-11 by dlech

jononi, 2022-06-13:
I confirm that the issue was fixed. Thanks!