pybricks/support #666: [Question] How to make async functions?

MrCheatEugene, 2022-06-11:
**Question**
Hi. I need some way to make asynchonic functions. Is there a way to make them?
**Context**
I have a program written in MicroPython for Lego EV3, and I need to play sound file, and keep the UI working. For that, I need any way to make asynchronic functions.

I tried to import 'asyncio', but it seems like it is not included in Pybricks MicroPython:
```
Starting: brickrun --directory="/home/robot/project" "/home/robot/project/main.py"
----------
Traceback (most recent call last):
  File "/home/robot/project/main.py", line 10, in <module>
ImportError: no module named 'asyncio'
----------
Exited with error code 1.
```
Again, is there a way to do that? Or is there any alternative that can do the thing in my application?

dlech, 2022-06-11:
Please see https://github.com/pybricks/support/discussions/356 for some ideas.

MrCheatEugene, 2022-06-12:
> Please see #356 for some ideas.

I tried importing the required modules, but now I get this: 
```
Traceback (most recent call last):
  File "/home/robot/project/main.py", line 6, in <module>
ImportError: cannot import name blocking_wait
```
It seems like blocking_wait does not exist?

MrCheatEugene, 2022-06-12:
Also, whitout this module my function does not work. It is intended to play sound, but I don't hear anything. And yes, the volume is not at 0%.
Functions:
```
def wait_first(*tasks):
    yield from zip(*tasks)
    for t in tasks:
        t.close()
def play(file):
    speaker.play_file(file)
    try:
        while True:

            yield
    finally:
        print("task stopped")
    pass
while (True):
    buttonsList=ev3.buttons.pressed()
    if(Button.CENTER in buttonsList):
        wait_first([play(files[selectedINDEX])])
        time.sleep(0.250)# delay for buttons, so they won't be pressed a bunch of times
pass
```
Why is this happening?

dlech, 2022-06-12:
> tried importing the required modules, but now I get this:
> 
> ```
> Traceback (most recent call last):
>   File "/home/robot/project/main.py", line 6, in <module>
> ImportError: cannot import name blocking_wait
> ```
> 
> It seems like blocking_wait does not exist?

If you get this error, then you probably didn't copy it exactly.

```python 
from pybricks.tools import wait as blocking_wait
```

Will import the `wait` function but give it the name `blocking_wait` instead of `wait`.

dlech, 2022-06-12:
> Why is this happening?

This is creating a new generator function on each iteration of the while loop so instead of the old generator function running all the way through, it is just starting a new generator function that pauses at the first yield.

dlech, 2022-06-12:
> def play(file):
>     speaker.play_file(file)
>     try:
>         while True:
> 
>             yield
>     finally:
>         print("task stopped")
>     pass

There is also nothing to break out of the `while: True:` loop here, so event if the sound did play, it would get stuck at this point.

Also, the `speaker.play_file()` method is blocking, so some tricks are needed to make this work in a non-blocking way. See https://github.com/pybricks/support/issues/99.

dlech, 2022-06-12:
Here is a complete working example based on your code:

```python
#!/usr/bin/env pybricks-micropython

import _thread
from experimental_c import pthread_raise
from usignal import pthread_kill, SIGUSR2

from pybricks.hubs import EV3Brick
from pybricks.ev3devices import (Motor, TouchSensor, ColorSensor,
                                 InfraredSensor, UltrasonicSensor, GyroSensor)
from pybricks.parameters import Port, Stop, Direction, Button, Color
from pybricks.tools import wait as blocking_wait, StopWatch, DataLog
from pybricks.robotics import DriveBase
from pybricks.media.ev3dev import SoundFile, ImageFile

# This program requires LEGO EV3 MicroPython v2.0 or higher.
# Click "Open user guide" on the EV3 extension tab for more information.

# Create your objects here.
ev3 = EV3Brick()

def wait_first(*tasks):
    yield from zip(*tasks)
    for t in tasks:
        t.close()

def play(file):
    lock = _thread.allocate_lock()
    lock.acquire()

    def play_in_background_thread():
        ev3.speaker.play_file(file)
        lock.release()

    thread = _thread.start_new_thread(play_in_background_thread, ())

    try:
        while lock.locked():
            yield
    except GeneratorExit:
        pthread_raise(thread, SystemExit())
        pthread_kill(thread, SIGUSR2)

def wait_button_pressed(button):
    while button not in ev3.buttons.pressed():
        yield

def play_file_when_center_button_pressed():
    while True:
        yield from wait_button_pressed(Button.CENTER)
        yield from play(SoundFile.HELLO)

# main event loop

tasks = [play_file_when_center_button_pressed()]

while True:
    for t in tasks:
        next(t)
        # don't hog CPU
        blocking_wait(10)
```

MrCheatEugene, 2022-06-15:
> Here is a complete working example based on your code:
> 
> ```python
> #!/usr/bin/env pybricks-micropython
> 
> import _thread
> from experimental_c import pthread_raise
> from usignal import pthread_kill, SIGUSR2
> 
> from pybricks.hubs import EV3Brick
> from pybricks.ev3devices import (Motor, TouchSensor, ColorSensor,
>                                  InfraredSensor, UltrasonicSensor, GyroSensor)
> from pybricks.parameters import Port, Stop, Direction, Button, Color
> from pybricks.tools import wait as blocking_wait, StopWatch, DataLog
> from pybricks.robotics import DriveBase
> from pybricks.media.ev3dev import SoundFile, ImageFile
> 
> 
> # This program requires LEGO EV3 MicroPython v2.0 or higher.
> # Click "Open user guide" on the EV3 extension tab for more information.
> 
> 
> # Create your objects here.
> ev3 = EV3Brick()
> speaker_lock = _thread.allocate_lock()
> 
> def play(file):
>     # this will raise OSError if lock is already held (speaker in use)
>     speaker_lock.acquire(False)
> 
>     def play_in_background_thread():
>         try:
>             ev3.speaker.play_file(file)
>         finally:
>             # having release in a finally block ensure that lock is released
>             # even if playback is interrupted
>             speaker_lock.release()
> 
>     thread = _thread.start_new_thread(play_in_background_thread, ())
> 
>     # wait for background thread to finish
>     try:
>         while speaker_lock.locked():
>             yield
>     except GeneratorExit:
>         pthread_raise(thread, SystemExit())
>         pthread_kill(thread, SIGUSR2)
>         # REVISIT: probably should wait for lock to be released here
> 
> 
> def wait_button_pressed(button):
>     while button not in ev3.buttons.pressed():
>         yield
> 
> 
> def play_file_when_center_button_pressed():
>     while True:
>         yield from wait_button_pressed(Button.CENTER)
>         yield from play(SoundFile.HELLO)
> 
> 
> # main event loop
> 
> tasks = [play_file_when_center_button_pressed()]
> 
> while True:
>     # run each task until the next yield statement
>     for t in tasks:
>         next(t)
> 
>     # give the OS some time to run
>     blocking_wait(10)
> ```

Thanks! It works! But I also need to stop the task somehow, while it's running. Can I do that?

dlech, 2022-06-15:
Yes, generators can be canceled by calling the `close()` method on a generator (task) object. Doing this without crashing the main event loop can be a bit tricky though, so it would be helpful to know the bigger picture of what you are trying to accomplish.

MrCheatEugene, 2022-06-27:
Sorry for not replying in a long time, I was doing other stuff at the moment.
Anyway, thanks. I will try it asap.
I am trying to make an music player out of simple EV3, whitout any button modules, just the ev3. Don't ask why:)

dlech, 2022-06-27:
OK, I won't ask. https://www.youtube.com/watch?v=MysBpNJkY2o

MrCheatEugene, 2022-06-28:
That's a synthesizer, I am making a music player, that plays music from .wav files on ev3, without any external buttons or modules. 
But, still a nice project!

dlech, 2023-03-21:
Future visitors will be interested in the latest developments #917.

closed 2023-03-21 by dlech